riscv-non-isa/riscv-trace-spec #10: Exceptions/interrupts vs. traps

pdonahue-ventana, 2019-05-17:
[RISC-V](https://github.com/riscv/riscv-isa-manual/releases/download/draft-20190515-2ff4f57/riscv-spec.pdf) has these definitions (which differ from other architectures I've used):

- Exception: "an unusual condition occurring at run time associated with an instruction in the current RISC-V hart"
- Interrupt: "an external asynchronous event that may cause a RISC-V hart to experience an unexpected transfer of control"
- Trap: "the transfer of control to a trap handler caused by either an exception or an interrupt"

Not all exceptions and interrupts cause traps.  Most notably, floating point exceptions and disabled interrupts do not trap.

If an exception or interrupt doesn't trap, the PC doesn't change.  There's no need to trace all exceptions/interrupts, just traps.  I'm sure that the trace architecture intends to only trace traps.

However, the trace spec has many references to tracing "exceptions" and/or "interrupts" but it should say "traps" to be consistent with the RISC-V terminology.

gajinderpanesar, 2019-05-22:
Paul,

  Thanks for this. I will address this in the next draft.
  Regards,

    Gadge

gajinderpanesar, 2019-07-09:
This has been address in the latest draft specification. We added Paul's clarification to the Nomenclature section.

closed 2019-07-09 by gajinderpanesar