riscv-non-isa/riscv-trace-spec #45: tvalepc for illegal instruction

weiqi7777, 2020-10-27:
hi:
from table 7.2, the tvalepc description:

> **Exception address if ecause is 2** and interrupt is 0(illegal instruction exception), or trap value otherwise

but for section 7.3.1.

> This field reports the address of illegal instructions, or the trap value otherwise. This ensures
that the address of the faulting instruction is reported for all required cases. The trap value is
set to the address of the faulting instruction for hardware breakpoints, access or page faults and
instructions, loads or stores that are mis-aligned, **but not for illegal instructions (for which it is set
to the opcode)**

the two descriptions are not match for illegal instruction.   
which description is correct?

thanks
lujun

IainCRobertson, 2020-10-27:
Lujun,
             The descriptions are correct and consistent.  The ‘trap value’ in both sections refers to the tval input provided by the HART.  “This field” in section 7.3.1 is tvalepc.

Disclaimer

The information contained in this communication from the sender is confidential. It is intended solely for use by the recipient and others authorized to receive it. If you are not the recipient, you are hereby notified that any disclosure, copying, distribution or taking action in relation of the contents of this information is strictly prohibited and may be unlawful.

This email has been scanned for viruses and malware, and may have been automatically archived by Mimecast Ltd, an innovator in Software as a Service (SaaS) for business. Providing a safer and more useful place for your human generated data. Specializing in; Security, archiving and compliance. To find out more visit the Mimecast website.

weiqi7777, 2020-11-03:
thanks for yout replay

closed 2020-11-03 by weiqi7777