riscv-non-isa/riscv-trace-spec #46: which variable of te_instruction_record_t meas "retire"?

zjuwlh, 2020-10-29:
Asking this, because in Instruction deta trace algorithm diagram in spec, if next inst is exc_only , will sent te_inst : format0/1/2.
but in encoder-algorithm-public.c line 1085, for "if ***" I can't see the exception without retirement information or condition.
also, if a uninferable jump instruction following a exception with retirement, what kinds te_inst will the c encoder send out?
as we know, it should send out te_inst format 2 following te_inst format 3 sub1, but actually , the c encoder send out two te_inst format 2 following te_inst format 3 sub1, could you please help check it?

Thanks,
Lihua

zjuwlh, 2020-10-29:
Another question is , in hart interface, there is a signal "itype", then what value of itype can be used to indicate te_instruction_record_t.is_updiscon=1?

gajinderpanesar, 2022-03-15:
The interface supports signalling an exception with or without retiring the instruction which caused it. The reference flow should have cases that cover this.