riscv-non-isa/riscv-trace-spec #60: trigger[2:0] behaviors for exception

zhonghochen, 2021-09-06:
Hi, Guys

https://github.com/riscv/riscv-trace-spec/blob/9896e28a5b35d86178c08e77e0ae6daaafeba072/ingressPort.tex#L492-L493

Could you help to clarify whether trigger[2:0] should be generated for non-retired instruction?
* mcontrol.execute is set, 
* An instruction matches the trigger, and
* The instruction causes an exception (e.g. ecall, illegal instruction, and address misaligned)

Regards,
Zhong-Ho

IainCRobertson, 2021-09-06:
Zhong-Ho,
                   I think it depends how you want to trace the instruction.  Do you want to start tracing from when this instruction is retired, or do you also want to trace the exception handler?

If you want to include the exception handler in the trace, then assert trigger[0] when the exception is reported.

If you only want to trace from when the instruction retires, then only assert trigger[0] if the instruction actually retires (many instructions that cause an exception will be re-executed and retired after the exception handler has completed).

(Note that ecall retires and generates an exception)

Iain

Iain Robertson, Engineering Director, Hardware
Tessent Silicon Lifecycle Solutions, Siemens EDA
***@***.******@***.***> | mobile: +44 (0)7743 369213
siemens.com/eda

[A picture containing text, clock  Description automatically generated]

https://github.com/riscv/riscv-trace-spec/blob/9896e28a5b35d86178c08e77e0ae6daaafeba072/ingressPort.tex#L492-L493

zhonghochen, 2021-09-06:
Hi, Iain,

Thanks for your replies. Since the behavior is dependent on usage, do you think the timing field in mcontrol can be used to control the behavior of trigger[2:0] ?

Regards,
Zhong-Ho

IainCRobertson, 2021-09-06:
Zhong-Ho,
                   I’m going to have to defer that question to someone more familiar with the Debug Module than myself.

Iain

pdonahue-ventana, 2021-09-10:
That behavior is a property of the trigger module, not trace.  Trace does whatever the trigger module tells it to do.

I opened https://github.com/riscv/riscv-debug-spec/issues/670 about this.  There I gave what I'm reasonably sure is the correct answer but we'll see if there are any other opinions.  That debug spec issue will make sure that the spec gets clarified by the debug task group (which will likely be done by me).