riscv-non-isa/riscv-trace-spec #79: Report timing of context

hwliu-tommy, 2022-07-21:
Hi,

I have 2 questions about the report timing of context.

If the context value (comes from CSR) is updated by a CSRRW instruction, which instruction should report the new context?
1. On the CSRRW instruction, report the next value of context (value from CSR write data)
2. On the next instruction, report the current value of context

And the other question is that can the context be reported with a block that only contains exception or interrupt?

Thanks,
Tommy

IainCRobertson, 2022-07-21:
Tommy,
              Responses inline

[Iain] I would expect the context to be reported on the next instruction (case 2), as the next context is in force from this point.

[Iain] Yes.  An exception or interrupt always results in a format 3, subformat 1 (sync-trap) packet, which includes context.

hwliu-tommy, 2022-07-21:
Hi, Iain,

Thanks for your reply.
It is very clear for question 1.

Sorry I didn't describe question 2 clearly.
The question is asking about the core to encoder interface.
Can the core reports context to encoder when itype=interrupt/exception, and iretire=0?

Since the spec says 
> Context for all instructions retired on this cycle. (context definition)
> Report the address of the 1st instruction retired in this block, and the new context. (from ctype = report precisely)

My interpretation is that the context change can only be reported when the next instruction is retired.
If the next instruction raises an exception and not being retired, the context won't be reported.

So there will be a corner case when an exception happens right after a context change.
For the following sequence:
```
1. CSRRW set context to 0 (itype=0, iretire=2)
2. exception taken, set context to 1 (itype=1, iretire=0)
3. instruction retired with context = 1 (itype=0, iretire=2)
```
If the context can only be reported on instruction retire, the context information before the exception taken might not be observed through trace.

Thanks,
Tommy

IainCRobertson, 2022-07-22:
Tommy,
               Thanks, I understand now.

From the interface point of view, the context value is not ignored in the cycle when itype = 1 or 2 and iretire=0.  However, in most cases, the context input on that cycle is ignored.  Under normal circumstances when an interrupt or exception occurs, it results in sync-trap te_inst (format 3 subformat 1) with thaddr = 1, reporting the details of the next instruction to retire (trap hander address), and the context value included in that packet will be that supplied on the interface when the trap handler retires.  Only tval and ecause values are taken from the exception cycle (context is ignored).

If there are back to back interrupts or exceptions, then a sync-trap packet is generated with thaddr=0, and this reports the tval, and ecause from the interrupt/exception, and the address of the instruction which faulted for an exception.  Only in this case will the context from the exception cycle be used.

As long as the context value input on the exception/interrupt cycle is the context that is in force at that time then all should be well.  The various non-zero ctype options will generate packets to ensure a new context is reported.  But if the context change coincides with an exception or interrupt, coming into force from the start of the handler, then there is actually no need to drive ctype, as whenever a sync-start or sync-trap message is generated it will report the context anyway.  ctype provides a mechanism to report it when otherwise there would be no reason to generate a sync packet.

Hope this is helpful.

Iain

Iain Robertson, Engineering Director, Hardware
Tessent Silicon Lifecycle Solutions, Siemens EDA
***@***.******@***.***> | mobile: +44 (0)7743 369213
siemens.com/eda

[A picture containing text, clock  Description automatically generated]

hwliu-tommy, 2022-11-03:
Hi, Iain,

Thank for the reply.
The clarification is very clear and it resolves my question.
And sorry for my late reply.

Thanks,
Tommy

closed 2022-11-03 by hwliu-tommy