riscv-non-isa/riscv-trace-spec #81: Question on recover_status_fields

cetola, 2023-01-03:
Had a question come up regarding the decoder model:

https://github.com/riscv-non-isa/riscv-trace-spec/blob/3280acdae80c23e9bd15b418af293ff0dbb01bfa/referenceFlow/scripts/decoder_model.py#L324

Specifically for branch with address packet, why must te_inst.branches == 31?

andyduller, 2023-01-05:
It's a long time since I looked at this ;-)
I believe that check is trying to differentiate between the two types of format 1 packet (format_t.BRANCH), described in Table 7.6 and Table 7.7. As suggested, it looks like this check is incorrect because branches == 0 means "no address is present".
The suggested fix of changing this to:
  (te_inst.format == format_t.BRANCH) and te_inst.branches != 0
would seem to be correct.

Apologies that this wasn't picked up in my testing.

cetola, 2023-01-06:
Thanks @andyduller, that was my guess, that te_inst.branches != 0 would make more sense.

Would you like me to have a PR sent in or will you handle this?

gajinderpanesar, 2023-01-09:
I have made the change so that we are now using  te_inst.branches != 0. So closing this issue

closed 2023-01-09 by gajinderpanesar