riscv-non-isa/riscv-trace-spec #92: How to handle cm_popret while decoding?

summorey22, 2024-01-21:


IainCRobertson, 2024-01-22:
You haven't given any detail, so I don't know exactly what your concern is here, but it should be as simple as including cm_popret as one of the opcodes that causes the 'is_inferable_jump' function in the decoder pseudo code to return true.

summorey22, 2024-01-22:
Got it. 

I have recently started exploring this topic and I am still studying it. That's why I didn't have a clear idea of how to classify this particular instruction.

According to the pseudocode for inferable jump, we would add the immediate bits in the instruction to the current pc to determine next pc. Is that enough to determine the next pc, or would it require the encoder to give an extra message?

IainCRobertson, 2024-01-22:
Sorry - a small typo totally changed my intended meaning and has sent you off in the wrong direction.  popret is an uninferable jump, and the affected function is 'is_**un**inferable jump'.  The program returns to the value stored in ra, which is popped from the stack.  The decoder cannot infer this value.  The encoder will output a packet containing the target address of this return.

closed 2024-03-05 by summorey22