riscv-non-isa/riscv-trace-spec #98: A vector of sideband signals to inform the trace output being stopped

AoteJin, 2024-04-09:
Based on discussion on external debug security TG meeting, adding another bit alongside halted signal to bundle a vector is the desired way to inform the trace to stop output due to different reasons. The reserved encoding is for futural extension.

[new_bit, halted] | Tracing? | Stop reason
-- | -- | --
00 | Yes | n/a
01 | No | entering Debug Mode
10 | No | due to security rules
11 | No | reserved for future

IainCRobertson, 2024-04-09:
I propose we call this signal ''notrace".

bcstrongx, 2024-06-04:
Yeah, seems like bit 0 shouldn't be called halted anymore, since for the reserved 0b11 case we may not be halted.  I might lean towards "traceinh[1:0]", but don't care that much.

One trick with this is that an implementation can't simply connect bit 0 to halted and bit 1 to security check failure.  Otherwise the encoder could see 0b11 when the hart is both halted and security checks fail (e.g., we halted in M-mode but trace is not allowed in M-mode).  The table above is clear enough, but the external debug spec probably needs clarification here.

IainCRobertson, 2024-06-04:
Agree - we need a new name for the new vector, not a new signal that has an encoded meaning with halted. That’s a recipe for screw ups.

I guess you’re thinking “trace inhibit”?  I don’t think we need “trace” in the name of the ingress port signal - they’re all for trace!  So we could just go with “inhibit”?  I prefer that to traceinh as it’s not so obvious to the casual reader what “inh” means.

bcstrongx, 2024-06-04:
good point, inhibit works

joxie, 2025-01-17:
Ping @IainCRobertson do we have a plan to add this to trace spec? We (nVidia team) can help to create a PR if needed.

IainCRobertson, 2025-01-17:
Not a detailed one.  In my view it should be part of the set of MRs submitted for ratification as part of the External Debug Security spec.  E-Trace is a ratified standard, so this would be a normative change that will require significant scrutiny and sign-off.  Needs to be properly coordinated.  I'm happy to create the content but you need to agree the details of how to go about this with ARC beforehand.

joxie, 2025-01-17:
Thanks for the response Iain. Agreed that it would be significant work since e-trace is ratified, I didn't realized that before. Feels like we can just use debug sec. spec to document the change, in that case we just need you to review, are you fine with that?

IainCRobertson, 2025-01-17:
Happy to review.  Work-around with existing ratified E-trace spec is to OR the signal that inhibits trace (I forget what it's called off the top of my head) into the HALTED ingress port signal.  This at least should be mentioned in the External Debug Security spec.

AoteJin, 2025-02-11:
Hi @IainCRobertson,

 I recall that there is ongoing work (probably a fast-track) to split ingress port definition to a separate spec. Is it going to be completed shortly? I am thinking about when we define the inhibit signal, whether it should refer to E-trace spec or a dedicated spec as a complementary signal.

IainCRobertson, 2025-02-11:
Aote,
          You are correct: there is a fast-track to split out the hart-trace interface (HTI).  It is approved but has not started yet.  I'm waiting on Rafael to set up the git repo.  Additionally I am in the process of preparing an update to the E-Trace spec (informative clarifications and corrections only) which is nearing completion.  My plan was to publish this first before starting on the work to split out the HTI, which will then require a second informative E-Trace spec update to remove the interface and reference the standalone doc.  Realistically I expect it to be a couple of months until the fast-track is done.

As part of the Secure Debug TG ratification I think you will need to raise a PR for the proposed change to the interface.  Once the HTI git repo is set up and populated, you should be able to raise this PR against the HTI repo.  I don't think you need to wait for it to be published.

Iain