riscv-non-isa/riscv-trace-spec #100: format error in newest E-Trace spec

zhangdujiao, 2024-04-18:
format error in "Table 24. Packet format 0, subformat 0 - address, branch count"
In some way, I cannot upload the snap.

And one question:

the bit column of irdepth is "return_stack_size_p (return_stack_size_p > 0 ? 1 : 0) call_counter_size_p" in Table 20. Packet format 2, which is different from the former version.

zhangdujiao, 2024-04-18:
and there is a typo in 7.6.2 , Bold below
"However,there are circumstances where this approach does not work. For example, consider the case where there is an exception at **looplabel 4**. In this case, the decoder cannot tell whether this occurred during the 1st or 2nd loop iterations, without additional information from the encoder. This is the purpose of the updiscon field. In more detail:"

IainCRobertson, 2024-04-19:
Thanks for pointing these out.  They were all introduced in the LaTeX to asciidoc translation that got missed during review.  

Column formating of table 24 is messed up - text in 1st column below notify row belongs in last column - accidental row break has been inserted.

The other two issues relate to + signs getting lost:
'looplabel 4' should read 'looplabel + 4'.
return_stack_size_p (return_stack_size_p > 0 ? 1 : 0) call_counter_size_p should be return_stack_size_p + (return_stack_size_p > 0 ? 1 : 0) + call_counter_size_p

@wmat can you address these as they were introduced during your transcoding?  Or would you prefer me to try?

wmat, 2024-04-19:
@IainCRobertson do you want a new 2.0.3 release on GitHub as well?

IainCRobertson, 2024-04-19:
Yes please - this needs to replace 2.0.2 everywhere.

wmat, 2024-04-19:
Done. Published version on Google Drive has also been updated by Jeff.

closed 2024-04-19 by wmat

IainCRobertson, 2024-04-19:
Thanks Bill :-)

zhangdujiao, 2024-04-20:
Thank you very much for your efficiency.

We are dedicated to developing the RISC-V trace IP. After research, we found that the E-Trace spec was ratified in 2022.05, and the E-Trace Encapsulation was recently frozen.

The current issue we are facing is whether the debugger software from Lauterbach or other company is compatible with the RISC-V Trace IP. After communicating with Lauterbach, they mentioned that once they confirm the E-Trace standard will not do any further modifications, they will start a support plan. 
So, we would like to ask if E-Trace will have any additional changes and supplements in terms of content?

ps. The N-Trace is currently not finalized, so we won't consider it for now. 

Thank you very much.
Best

IainCRobertson, 2024-04-22:
There are no committed plans for "additional changes and supplements" to E-Trace at this time, but as with all things there are several possibilities for future enhancements that may be considered in the future.  For example, extensions to support cycle accurate trace or performance profiling.  Any such extensions would need to be optional and 100% backwards compatible with the existing ratified spec, so there is no reason to be concerned here.

I think Lauterbach's concern about "further modifications" relates primarily to E-Trace Encapsulation, and as you are aware this is now frozen and very close to being formally ratified.

Iain

zhangdujiao, 2024-04-22:
Many Thanks for your reply! It helps a lot!

Best