riscv-non-isa/riscv-trace-spec #103: The specific implementation principle of the implicit return mode

zhangdujiao, 2024-04-25:
Every time a call instruction is encountered, the predicted return address (obtained through iaddr, iretire, and ilastsize) is pushed onto the stack. Every time a return instruction is encountered, it is matched with the predicted address on the stack, and then it is determined whether the prediction is correct. If it is correct, no packet is sent. If it is incorrect, a packet containing the address is sent, right?

These paragraphs below in 7.6.3. Format 2 irreport and irdepth are not very well understood:

"In some cases it is also necessary to report the current stack depth or call count if the packet is reporting the last instruction before an exception, interrupt, privilege change or resync. There are two cases of concern:
• If the reported address is the instruction following a return, and it is not mis-predicted, the encoder must report the current stack depth or call count if it is non-zero. Without this, the decoder would attempt to follow the execution path until it encountered the reported address from the outermost nested call;
................
• If the reported address is not the instruction following a return, the encoder must report the current stack depth or call count unless:
◦ There have been no returns since the last call (in which case the decoder will correctly stop in the innermost call), or
◦ There has been at least one branch since the last return (in which case the decoder will correctly stop in the call where there are no unprocessed branches)."

Can you help me make it clear? It's a little bit hard for me to understand this from the Decoder point of view.

pdonahue-ventana, 2025-01-03:
@IainCRobertson: I have a similar question about 7.6.3 which says:
> In some cases it is also necessary to report the current stack depth or call count if the packet is
reporting the last instruction before an exception, interrupt, privilege change or resync. There are two
cases of concern

Does this mean that the stack depth is reported only in the two listed cases or does this mean that it's always reported before an exception, interrupt, privilege change or resync and the two cases that follow are just cases of concern that are illustrative of why the general requirement is necessary?

IainCRobertson, 2025-01-06:
It must be reported in those 2 listed cases.  For other cases there is no need but it would be harmless (apart from the reduction in overall encoding efficiency).

closed 2025-01-08 by zhangdujiao