riscv-non-isa/riscv-trace-spec #106: How to define the newest valid group for R type signal

zhangdujiao, 2024-05-06:
How to understand the spec below from chapter 4.2 trace spec.
"For harts that can retire a maximum of N non-zero itype values per clock cycle, the signal groups MR, OR and either BR or SR must be replicated N times. Typically N is determined by the maximum number of branches that can be retired per clock cycle. Signal group 0 represents information about the oldest instruction block, and group N-1 represents the newest instruction block. The interface supports no more than one privilege change, context change, exception or interrupt per cycle and so signals in groups M and O are not replicated. Furthermore, itype can only take the value 1 or 2 in one of the signal groups, and this must be **the newest valid group** (i.e.
iretire and itype must be zero for higher numbered groups). If fewer than N groups are required in a cycle, then lower numbered groups must be used first. For example, if there is one branch, use only group 0, if there are two branches, instructions up to the 1st branch must be reported in group 0 and instructions up to the 2nd branch must be reported in group 1 and so on."

what does the newest valid group mean? For example, our CPU can retire 2 special instruction (itype≠0) simultaneously, that is group0 and group1 are available for valid signal. 
Signal group 0 represents information about the oldest instruction block, and group N-1 represents the newest instruction block.
For itype=1 or 2, will there be a case where group1 is the newest valid group? 
Or for itype=1 or 2, only group0 is available？ and iretire and itype in group1 must be zero. Since "The interface supports no more than one privilege change, context change, exception or interrupt per cycle and so signals in groups M and O are not replicated."
While the spec "Signal group 0 represents information about the oldest instruction block, and group N-1 represents the newest instruction block." made me confused..

IainCRobertson, 2024-05-08:
See inline…

Iain

[Iain] Correct.

[Iain] I can’t speak for your particular CPU, but in general this is possible.  For example, you could retire a block of instructions in group 0, and report an exception or interrupt in group 1.  What isn’t allowed is to have itype = 1 or 2 in group 0 and retire more instructions in group 1.  If itype is 1 or 2 it must be in the highest numbered valid group (see green highlight above).

[Iain] This doesn’t mean you can only use one group when there is a privilege change, context change, exception or interrupt.  Hopefully the example above clarifies this.

zhangdujiao, 2024-05-11:
> or example, you could retire a block of instructions in group 0, and report an exception or interrupt in group 1. What isn’t allowed is to have itype = 1 or 2 in group 0 and retire more instructions [...]

Does this mean that if there is an itype=1 or 2 in group0, there is only one valid group0, and group1 or higher has no valid signal?
thanks!

raulbehl, 2024-05-12:
> Does this mean that if there is an itype=1 or 2 in group0, there is only one valid group0, and group1 or higher has no valid signal?

Yes, that is correct. You can only have group0 valid if itype=1 or 2.

IainCRobertson, 2024-05-16:
Yes, exactly.

Iain

closed 2024-05-23 by zhangdujiao