riscv-non-isa/riscv-trace-spec #109: How to understand exc_only and er_n in Chapter 9

zhangdujiao, 2024-05-15:
exc_only: Exception or interrupt signalled without simultaneous retirement.
Does this mean that current block only contains exception or interrupt, and no other retired instruction? i.e. itype=1/2 and iretire=0?

er_n: Instruction retirement and exception signalled on the same cycle, or Trace notify trigger.
Does this mean that current block contains exception or interrupt and also other retired instructions? i.e. itype=1/2 and iretire≠0?

thanks!

IainCRobertson, 2024-05-16:
Yes (to both questions)

Iain

IainCRobertson, 2024-05-20:
> Yes (to both questions)
>
> The instruction retired before interrupt or exception must be reported, from"When an interrupt or exception occurs, the final instruction retired beforehand must be traced.", Chapter 3.1.4.
>
> For the condition of "er_n", there are no less than one retired instruction before interrupt or exception, like the retired block below:
> inst.1
> inst.2
> ...
> inst.n
> exception.
> [itype=1, iretire=N]
> In this case, how to report the address of inst.n before the exception?
>
> For the case of "exc_only" block below:
> --itype=0, iretire=N
> inst.1
> inst.2
> ...
> inst.n
>
> --itype=1, iretire=0
> exception.
>
> block1 satisfies the in Figure2 Instruction delta trace algorithm, Chapter9. So, the address before the exception (addr of inst.n) can be reported.
> But, what for the case of "er_n"? by what condition the TE can report the address before except. of int.?
>
> Many Thanks!
>
> -

I don't see a problem here.  When a block of instructions retires, you know:

  *   The address of the 1st instruction in the block
  *   The number of half-words in the block (iretires)
  *   The size of the final instruction (lastsize)

From this you can compute the address of the last instruction retired in the block.  Regarding the er_n case vs exc_only, the only difference is whether you use the last address from the previous block (exc_only) or the current block (er_n).

Iain

closed 2024-05-21 by zhangdujiao