riscv-non-isa/riscv-trace-spec #112: There is no SRCID filed in ATB Encapsulation

zhangdujiao, 2024-05-31:
From the description from Chapter 7, ATB Encapsulation contains: 
• A 5-bit field specifying the payload length in bytes
• A bit to indicate whether an optional 16-bit timestamp is present;
• An optional 2-byte timestamp;
• The packet payload.
The encapsulation has no SRCID filed. While the ATID is only 7bit width, witch may not be enough, How to solve this problem?

thanks!

IainCRobertson, 2024-06-02:
You should really be working from the E-Trace Encapsulation spec, rather than Chapter 7 of the E-Trace spec.  The latter is only illustrative, and not part of the standard.  The E-Trace Encapsulation TG was set up specifically to produce a formalized standard for the encapsulation.

As noted previously, the length of the timestamp can be any number of bytes and is not limited to 16-bits.

In terms of ATID being only 7 bits, I think the way that is usually handled (in Arm-based systems) is to dynamically allocate the SrcID/ATID at run-time.

Iain

zhangdujiao, 2024-06-03:
Thanks Iain, Does this mean that if the ATB encapsulation is used, then the 
ATDATA = {header (8 bit), srcID (0-16 bit), timestamp (64bit max), payload (248bit max)}, from <Unformatted Trace & Diagnostic Data Packet Encapsulation for RISC-V>.
If we need more than 128 cores can be traced simultaneously, srcID is necessary since ATID can not be enough to illustrate every source of trace flow.

IainCRobertson, 2024-06-03:
Yes, or you could use a hybrid approach where the 7LSBs of the SrcID are carried by ATID and the remainder are included in the packet encapsulation.

Iain

zhangdujiao, 2024-06-04:
> 7LSBs of the SrcID are carried by ATID

In this case, the ATID may be repeated in SoC, and the components in SoC, like Funnel, ETR processe the flows carrying the same ATID as one flow.

IainCRobertson, 2024-06-04:
Possibly.  I'm not an expert in ATB so you may want to consult someone else on this, and the ways in which the 7-bit ATID limit is handled.

Iain

zhangdujiao, 2024-06-04:
Thanks Iain

closed 2024-06-04 by zhangdujiao