riscv-non-isa/riscv-trace-spec #114: Payload size is limited in 31bytes (248bit)

zhangdujiao, 2024-06-05:


zhangdujiao, 2024-06-05:
The max size of Format 3 subformat 1 - Trap is 330 bit theoretically.
<Unformatted Trace & Diagnostic Data Packet Encapsulation for RISC-V> limit the payload size in 248 bit. Do we need to send two packets separately?

IainCRobertson, 2024-06-05:
I'd be interested to see how you got to 330 bits.  But yes, if the payload exceeds 248 bits it would need to be split into 2.  There are various ways this could be accommodated, but I don't think it should be necessary as I don't believe any real-world system is going to get anywhere near the 248-bit limit.

Iain

zhangdujiao, 2024-06-05:
> I'd be interested to see how you got to 330 bits.

For example:
Format 3 subformat 1 - Trap
"
format 2bit
subformat 2bit
branch 1bit
privilege 3bit
time 64 bit
context 64 bit
ecause 64 bit
interrupt 1 bit
thaddr 1bit
address 64bit
tval 64bit
"
The payload size 330bits total.
We can put timestamp in ATDATA rather than payload, while the payload is 266 bits then, still larger than 248 bits.

zhangdujiao, 2024-06-06:
> I have never seen any suggestion that ecause is more than 6 bits wide. Currently any codes > 64 are reserved, and even if this changes at some point the total number of bits might increase to 7 or 8, but not 64 (actually 63, as the MSB of the xcause registers is used for the interrupt bit).

Got it, we need to re-considerate the width of port signals, many thanks Iain.

zhangdujiao, 2024-06-07:
> the MSB of the xcause registers is used for the interrupt bit

The ingress port signal **cause** includes the value from ucause/scause/ vscause/mcause, and the MSB of the xcause registers is used for the interrupt bit. So, the whole xcause value should be included in the packet. Otherwise, the decoder cannot determine the cause as an interrupt or an exception, right?

IainCRobertson, 2024-06-07:
Yes.  But you only need as many bits in ecause as there are defined exception causes.  This is currently a max of 6 bits, and in some implementations only 5.

Iain

zhangdujiao, 2024-06-07:
> But you only need as many bits in ecause as there are defined exception causes. This is currently a max of 6 bits, and in some implementations only 5.

If only the LSB 6 bits of xcause are contained in packet, instead of all 64 bits, can we determine whether the xcause value indicates an interrupt or an exception cause? Can the **interrupt field** in Trap packet indicate an interrupt or exception?

IainCRobertson, 2024-06-07:
Yes, because the MSB is connected to the 'interrupt' signal in the interface.

Iain

zhangdujiao, 2024-06-07:
> Yes, because the MSB is connected to the 'interrupt' signal in the interface.

Got it! Many thanks!

closed 2024-06-07 by zhangdujiao