riscv-non-isa/riscv-trace-spec #115: Trace trigger hit when triggers with other action also fire

AoteJin, 2024-06-07:
The debug spec says:
_"What happens with trace actions when triggers with different actions are also firing is left to the trace specification."_

But I didn't find any related description in the e-trace spec.

Considering the case, a icount trigger is set to enter debug mode when a given PC 0x1234 is executed 100 times. Meantime, another trigger's action is configured to be trace-on at PC 0x1234. In this case, neither of the trigger should silence the other one and both should take effective. Otherwise, the user will witness one of trigger behaves as disabled. 

 I think we need to add clear definition to cover it.

IainCRobertson, 2024-06-07:
There was a long discussion about this on the DTPM mailing list recently (subject was "Multiple trigger hit with different action" ). Issue #99 was generated in response to this.  I believe this addresses your concern.

AoteJin, 2024-07-25:
I went through the mail thread and it is addressing the action regarding trace on/off/notify. Actually what I am concerned about is the case when trigger set to action=1 ( enter debug mode) fires at the same time when another trigger set to trace related actions. 

I would assume that the trace action (non-intrusive debug) should not suppress debug mode (intrusive debug) or vice verse. 

Probably either debug spec or trace spec should elaborate on it?

IainCRobertson, 2024-07-25:
That's a debug spec question, but I'd assume not also.

pdonahue-ventana, 2024-07-25:
The debug spec says:
> What happens with trace actions when triggers with different actions are also firing is left to the trace specification.

This sentence was added in 2018 and is unchanged from the [Debug 0.13 spec](https://riscv.org/wp-content/uploads/2019/03/riscv-debug-release.pdf) that was ratified over 5 years ago (before E-Trace was ratified).  It looks like there might have been a lack of coordination here.

AoteJin, 2024-07-26:
@pdonahue-ventana  do you suggest to open an issue in debug repo and close this one?

pdonahue-ventana, 2024-07-26:
It's a strange and unfortunate situation.  The debug spec came first and didn't know what trace wanted so it said that trace would answer the question.  Debug got ratified.  Trace didn't fill in the details and that got ratified without anybody noticing.  The two options are to (1) put in some post-ratification clarification in the trace spec or (2) put something in the frozen but not quite ratified debug 1.0.  Although the first option would also provide compatibility with debug 0.13, I don't think that we care about that in 2024.  Option 2 makes the most sense.  I assume that @IainCRobertson agrees.

As for the actual text that gets added, I think that there should be joint agreement between the trace and debug people.  We don't want debug to dictate something that is bad for trace and vice-versa.

@rtwfroody

IainCRobertson, 2024-07-26:
Actually, the statement Paul quoted from the Debug spec:

"What happens with trace actions when triggers with different actions are also firing is left to the trace specification"

Was not ignored or overlooked in the Trace specification.  But the trace specification only addresses this in the context of triggers affecting trace: what happens if trace-on and trace-off occur together.  It does not say anything about what happens to the debug functionality if a trigger directed towards debug happens at the same time as a trigger directed towards trace, and nor should it in my view.   The debug functionality is described in the debug spec, and any clarification regarding how it responds to triggers originating from within the debug module has to be defined there.

Trace is essentially a passive functionality - the whole point is that it is non-intrusive.  There is no mechanism defined in the debug spec or trace spec for trace to affect the behaviour of the debug module.

Iain

AoteJin, 2024-07-31:
Intuitively, I agree on that the trace behavior should not affect the debug behavior. Since trigger itself is a module defined by debug spec, probably it is more adequate to elaborate on this case in scope of debug spec?

rtwfroody, 2024-08-02:
Ideally we'd have a statement like "Triggers with debug actions (0, 1) exist completely independently of triggers with trace actions (2, 3, 4, 5). They have no effect on each other."

Questions:
1. Is that backwards compatible?
2. Do we need something similar for triggers with external trigger output actions (8, 9)?

pdonahue-ventana, 2024-08-28:
@rtwfroody: Maybe a recommendation like this?
> Triggers with actions in each of these three categories should exist completely independently of triggers with actions in each of the other categories:
> - debug actions (0, 1)
> - trace actions (2, 3, 4, 5)
> - external trigger output actions (8, 9)

Maybe actions 8 and 9 should even be independent of each other (making four categories) since you could signal both triggers at the same time.  You can't trap and halt at the same time and you can't turn trace on and off and do a trace-notify at the same time.

rtwfroody, 2024-08-29:
Or maybe we should just say that actions 0 and 1 are specified in our priority table. All others may fire simultaneously. I know the trace spec had a discussion about what it means when multiple ones fire at the same time, so they've already addressed that.

pdonahue-ventana, 2024-08-29:
That's good.

IainCRobertson, 2024-10-24:
This issue has had a lot of discussion, with some aspects being addressed by updates to the Debug spec, and the E-Trace clarifications separated out into Issue #99.  The updates for this are included in https://github.com/riscv-non-isa/riscv-trace-spec/pull/158.  Any objections to closing this issue @pdonahue-ventana @AoteJin ?

AoteJin, 2024-10-24:
It looks great to me. Thanks for the heads up!

AoteJin, 2025-02-12:
Close since there is no objection.

closed 2025-02-12 by AoteJin