riscv-non-isa/riscv-trace-spec #116: The difference between encoder_mode and ioptions fileds in Format 3 subformat 3 - Support

zhangdujiao, 2024-06-11:
The width of encoder_mode and ioptions is both "N".
It seems that they all indicate the trace mode of the encoder. What is the difference between them?
The encoder_mode parameter in the configuration file indicates that only value 0 is supported, which means Branch trace mode. How should this mode be understood? Is it not the BTM mode in N-Trace?

IainCRobertson, 2024-06-11:
That's a poor choice of variable name that should have been spotted before.  The two 'N' are not at all related.  Allowing encoder_mode to be more than 1 bit is an attempt to future-proof for alternative or extended trace modes.  The E-Trace spec currently only defines one mode (branch trace), hence this will be a single bit set to 0 for any implementation that just implements this standard.

Iain

zhangdujiao, 2024-06-12:
Got it, thanks!

closed 2024-06-12 by zhangdujiao