riscv-non-isa/riscv-trace-spec #119: Encoder reference python script generatedm 3 TRAPs on 2 exception events

yandumchin, 2024-06-16:
Hi, 
I ran refernce encoder python script with the following inputs: 
0,0,0,0,0x80000104,4,0,0,0,0,0 
0,0,0,0,0x80000108,4,0,0,0,0,0 
10,0,0,0,0x8000010C,4,0,0,0,0,0  - updiscon
1,6,0xEE,0,0x8000222,0,0,0,0,0,0  - exception
2,0,0,0,0x800001b0,0,0,0,0,0,0 - interrupt
0,0,0,0,0x800001e0,4,0,0,0,0,0 
0,0,0,0,0x800001e4,4,0,0,0,0,0 
0,0,0,0,0x800001e8,4,0,0,0,0,0

and got the following outputs: 
**format=SYNC, subformat=SUPPORT, ienable=1, encoder_mode=0, ioptions=23, qual_status=0, denable=0, dloss=0, doptions=0 Payload[1f 17]**  
_next=80000154 next=80000104 curr=80000154_ 
**format=SYNC, subformat=START, address=80000154, branch=1, context=0, privilege=0, time=0, Reason[first_qualified] Payload[13 00 00 00 00 00 00 00 00 aa 00 00 40]** 
_next=80000108 curr=80000104 prev=80000154_ 
**format=ADDR, address=80000104, irreport=0, irdepth=511, notify=0, updiscon=0, Reason[prev_updiscon] Payload[12 04 00 00 02 00 00 00 e0]** 
_next=8000010c curr=80000108 prev=80000104 
next=8000222 curr=8000010c prev=80000108_ 
**format=ADDR, address=8000010c, irreport=0, irdepth=511, notify=0, updiscon=0, Reason[next_exc_only] Payload[32 04 00 00 02 00 00 00 e0]** 
_next=800001b0 curr=8000222 prev=8000010c_ 
**format=SYNC, subformat=TRAP, address=8000222, branch=1, context=0, ecause=6, interrupt=0, privilege=0, time=0, thaddr=0, tval=ee, Reason[prev_updiscon, curr_exc_only] Payload[17 00 00 00 00 00 00 00 00 83 88 00 00 02 00 00 00 80 3b]** 
_next=800001e0 curr=800001b0 prev=8000222_ 
**format=SYNC, subformat=TRAP, address=800001b0, branch=1, context=0, ecause=6, interrupt=0, privilege=0, time=0, thaddr=0, tval=ee, Reason[prev_exception, curr_exc_only] Payload[17 00 00 00 00 00 00 00 00 03 6c 00 00 20 00 00 00 80 3b]** 
_next=800001e4 curr=800001e0 prev=800001b0_ 
**format=SYNC, subformat=TRAP, address=800001e0, branch=1, context=0, ecause=0, interrupt=1, privilege=0, time=0, thaddr=1, Reason[prev_exception] Payload[17 00 00 00 00 00 00 00 00 30 78 00 00 20]** 
_next=800001e8 curr=800001e4 prev=800001e0_ 
**format=ADDR, address=800001e8, irreport=0, irdepth=511, notify=0, updiscon=0 Payload[a2 07 00 00 02 00 00 00 e0] format=SYNC, subformat=SUPPORT, ienable=0, encoder_mode=0, ioptions=23, qual_status=1, denable=0, dloss=0, doptions=0 Payload[4f 17]**

as you can see 3 TRAP packets are generated as response to uninferred jump and 2 exception events. I expected 2 TRAPs and SYNC_START instead.

IainCRobertson, 2024-06-17:
I agree - this looks like a bug in the reference encoder.

Iain

yandumchin, 2024-06-17:
Hi Lain,
Thanks for your quick response.
When encoder's fix is expected?

Yan.

IainCRobertson, 2024-06-17:
Yan,
        That's a good question.  The person who wrote the original encoder reference model is no longer involved with RISC-V, so we will need to find someone to take this on.

Iain