riscv-non-isa/riscv-trace-spec #120: 32 bits width of branch_count field in format 0, subformat 0

zhangdujiao, 2024-06-24:
For optional Branch prediction mode, if the conditions for sending a branch are met, and pbc ≥31, the TE will send format 0, subformat 0 to report branch_count rather than branch_map by Format 1. But the width of branch_count field is fixed at 32 bits, which is actually too large for the compress purpose, and branch_count field can even hardly overflow (pbc needs to be greater than 2^32+31).
Can the width be reduced to a reasonable value? What is the recommended value? 
The width in table is fixed to 32 bits. If we reduce it, whether can the software decode it?

IainCRobertson, 2024-06-24:
What's your concern here?  Packets with branch_fmt = 10 should be very rare, so the impact on encoding efficiency should be negligible.  There is currently no support for the branch_count width being less than 32.  If you reduce it, you could modify your decoder accordingly, but this would be non-compliant and wouldn't interoperate with other standard decoders.

Iain

zhangdujiao, 2024-06-24:
> What's your concern here?

My concern is 32bits width branch_count couldn't bring any compression optimization compared to without **branch prediction mode**, which should send Format 1 originally.

Format 1: 38 bits
Format 0 sub 0: 36/37 bits
branch prediction mode brings little bandwidth optimization.

IainCRobertson, 2024-06-24:
I disagree.  Most of the time, the upper bits of branch_count and all the bits above it will be zero and will compress away, resulting in a very short packet.

Iain

zhangdujiao, 2024-06-25:
oh I forget the compression method, thanks.

closed 2024-06-25 by zhangdujiao