riscv-non-isa/riscv-trace-spec #124: 1st qualified definition

zhangdujiao, 2024-06-25:
For figure 9.1, "qualified" means an inst. that meets the filtering criteria is qualified, and will be traced.
In the actual implementation, **can qualified be regarded as trace_on = 1?**

trace_on =  trTeActive && trTeEnable && trTeInstTracing && (!sec_check) && qualified
which trTeActive, trTeEnable and trTeInstTracing are configured by reg, sec_check is defined by 《RISC-V External Secure Debug v1.0》, qualified is derived from filter as defined above.

**Can "1st qualified" can be regarded as  trace_on 0->1?**

IainCRobertson, 2024-06-25:
Yes, if there is no filtering support, if filtering is disabled, or when filtering is enabled, if the first instruction for which trace_on is matches the filtering criteria.

Iain

zhangdujiao, 2024-06-27:
What if the instruction flow (t0~t3) below:
t0: unqualified
t1: qualified
t2: unqualified
t3: qualified
Can both t1 and t3 be regarded as 1st qualified?

IainCRobertson, 2024-06-27:
The support packet with qual_status = 0, reporting which trace features are enabled should be sent prior to the sync-start packet resulting from t1.  The spec doesn't say rexactly when but I would recommend sending it as soon as trace is enabled.

A support with qual_status = ended_* should be sent after the format 1 / 2 reporting the final traced instruction at t2.

There is no need to send a support packet associated with t3.  The triggering conditions are:

  *   Tracing ends
  *   Trace is lost
  *   The trace encoder is enabled
  *   The trace encoder configuration changes (and it is enabled after the change)

Iain

zhangdujiao, 2024-06-28:
thanks! Which made me clear on support packet trigger condition.
I am still confused on 1st qualified definition. For the trace on/off is controlled by filter simply, each time unqualified causes trace off, could the next qualified be considered as 1st qualified? and send the sync-start packet.

There is no need to send support packet and sync-start at t3?

> There is no need to send a support packet associated with t3.

IainCRobertson, 2024-06-28:
It simply means when tracing starts: instruction X is traced but instruction X-1 was not.  The objective is to give the decoder all the information it needs to be able to start decoding the trace.

Iain

zhangdujiao, 2024-06-29:
> t0: unqualified t1: qualified t2: unqualified t3: qualified

so for this condition,
t1 will issue support and sync-start, t2 will issue support due to trace off, and t3 will issue support and sync-start, am I right?

IainCRobertson, 2024-07-01:
You could do this, but it's not quite what I would do.  The point of the support packet is for two purposes:

  *   To tell the decoder how the encoder is configured and which optional modes are enabled
  *   To tell the decoder that tracing has ended

A support packet is not required to tell the decoder that tracing has started - the sync-start or sync-trap does this.  With this in mind, in your sequence, what is required is:

  *   Before t1: support with qual_status = 0 and enable = 1, to tell decoder how the encoder is configured.  You could send this immediately before the sync-start at t1, but it could also be sent earlier, when the encoder is enabled for example
  *   t2: I agree: support due to trace off (after format 1 / 2 reporting final traced instruction)
  *   t3: no support is needed here.  It's harmless, and allowed by the standard, but is a waste of bandwidth in my view.

Iain

zhangdujiao, 2024-07-01:
> * t3: no support is needed here. It's harmless, and allowed by the standard, but is a waste of bandwidth in my view.

OK I got it, the support packet only issues when the configuration of encoder is changed or trace has ended.
Could TE issue sync-start at t3? since trace-off at t2 and trace-on at t3

IainCRobertson, 2024-07-01:
Yes, it must issue sync-start or sync-trap at t3.

Iain

zhangdujiao, 2024-07-02:
got it, many thanks!

zhangdujiao, 2024-07-25:
> * Tracing ends * Trace is lost * The trace encoder is enabled * The trace encoder configuration changes (and it is enabled after the change)

Is the "The trace encoder is enabled" condition reasonable?
trace_enable = trTeActive && trTeEnable && trTeInstTracing, which the "(!sec_check) && qualified" are not included.

For the "Tracing ends", Is the following definition reasonable?
tracing end means (!trace_on),
trace_on = trTeActive && trTeEnable && trTeInstTracing && (!sec_check) && qualified

IainCRobertson, 2024-07-25:
See inline

Iain

[Iain] In my view the definition is correct.  It's reasonable to send this packet as soon as the Trace Encoder is capable of outputting trace, even if it is not yet doing so because of the filtering conditions (i.e. not qualified).  The purpose of this packet is to give the decoder the information it needs to correctly configure the decoder (which optional modes are enabled).  The sync-start packet will indicate when tracing actually starts.

[Iain] Yes.  I view sec_check as being factored into qualified so doesn't need to be called out explicitly in this statement.  Halted is also factored into qualified.  The important thing here is to tell the decoder that tracing has stopped and the final traced instruction has been reported.

-

zhangdujiao, 2024-07-29:
> Tracing ends

From the spec, the support packet is issued when:
-trace is enabled or disabled
etc. 
Trace is enable can be defined by trace_enable = trTeActive && trTeEnable && trTeInstTracing.
That is to say support will be issued even when secure check has not passed. Should !sec_check be added to trace_enable?

IainCRobertson, 2024-07-29:
Yes, but updating to include this will be handled as part of the ratification process for the Sdsec ISA extension.

Iain

closed 2024-08-01 by zhangdujiao