riscv-non-isa/riscv-trace-spec #125: Counter start condition of resynchronisation timer

zhangdujiao, 2024-06-26:
We can configure the resync timer mode by trTeControl.trTeInstSyncMode, and the max value by trTeControl.trTeInstSyncMax.
If we choose the inst. half word count. What is the time to start the Counter? When the trace start? and the qualified of filter should be 1 also right?

IainCRobertson, 2024-06-26:
It should start counting from the instruction which causes the 1st Sync-Start message, and should reset each time a sync-start or sync-trap is generated.  Same is true for any of the sync modes (it's just what you count that differs).

Iain

zhangdujiao, 2024-06-26:
> sync-start or sync-trap

what about sync-context and sync-support, they are both sync packet.
resync counter shouldn't work when trace is off, right?

zhangdujiao, 2024-06-26:
For unqualified instructions, whether to start the counter?
For example, whether to start the clk counter for unqualified inst.

IainCRobertson, 2024-06-26:
The point of the resync timer is to ensure that the decoder can find a starting point in the trace stream, and trace decoding can only start from sync-start or sync-trap.

An example of why the resync timer is required is when trace is written continuously to a circular buffer, over-writing the oldest data.  The buffer contents are only output when some triggering condition occurs, and in general the oldest packet in the buffer will not be a sync packet.  The decoder has to parse forward through the packets until it finds a sync-start or sync-trap packet in order to start decoding, and the resync timer puts an upper bound on how much trace the decoder will have to skip over before finding this.

Iain

IainCRobertson, 2024-06-26:
No, as unqualified instructions don't generate any trace.  Hopefully my answer to your previous question will give a bit more context.

Iain

zhangdujiao, 2024-06-27:
That helps a lot! thanks!

closed 2024-06-27 by zhangdujiao