riscv-non-isa/riscv-trace-spec #126: Why syncs-start packet is issued after two consecutive exceptions/interruptions?

zhangdujiao, 2024-06-28:
What are the principles for this packet sending mechanism?

IainCRobertson, 2024-06-28:
Normally, when there is a single trap, the following occurs:

  *   Format 1 or 2 indicating last instruction retired before the trap
  *   Sync-trap (with thaddr = 1) reports the cause of the trap, and the address of the 1st instruction that retires after the trap

Now consider a pair of consecutive traps.  The above case doesn't work, because the instruction after the 1st trap does not retire.  So instead we use the following sequence:

  *   Format 1 or 2 indicating last instruction retired before the trap
  *   Sync-trap with thaddr = 0 reports the cause of the 1st trap (and the address of the instruction which raised the exception).
  *   Sync-trap with thaddr = 1 reports the cause of the 2nd trap and the address of the 1st instruction that retires after the 2nd trap

There is another case related to exceptions at the targets of indirect jumps (updiscons).  In the usual case (the first one above), the instruction which caused the exception is known: it is the instruction which follows the one reported in the format 1 / 2.  However, if that instruction is an indirect jump, the decoder doesn't know the target address, and so is unable to determine which instruction caused the exception.  So when the exception occurs at the target of an uninferable jump, the following sequence is used:

  *   Format 1 or 2 indicating last instruction retired before the trap
  *   Sync-trap with thaddr = 0 reports the cause of the trap, and the address or the instruction which raised the exception
  *   Sync-start reports the address of the 1st instruction that retires after the trap

If a double-trap occurs after an uninferable jump you get

  *   Format 1 or 2 indicating last instruction retired before the trap
  *   Sync-trap with thaddr = 0 reports the cause of the 1st trap, and the address or the instruction which raised the exception
  *   Sync-trap with thaddr = 0 reports the cause of the 2st trap, and the address or the instruction which raised the exception
  *   Sync-start reports the address of the 1st instruction that retires after the 2nd trap

If there are more than 2 consecutive traps, the number of sync-trap packets with thadd=0 increases accordingly.

Iain

zhangdujiao, 2024-06-29:
Thanks Iain, great to have you! Made me have a deeper understanding of the mechanism.
still a little question for the quoted scene, it seems Sync-start will issue in the last from figure 9.1, since <reported?> is satisfied. So, the address of the 1st instruction that retires after the 2nd trap will be reported by sync-start packet, am I right?

> Now consider a pair of consecutive traps. The above case doesn't work, because the instruction after the 1st trap does not retire. So instead we use the following sequence: * Format 1 or 2 [...]

IainCRobertson, 2024-07-01:
Yes, exactly.

Iain

closed 2024-07-01 by zhangdujiao