riscv-non-isa/riscv-trace-spec #131: Trace mode configuration

zhangdujiao, 2024-07-23:
For the trace mode, like delta address mode, full address, implicit exception mode, etc..
Can we configure multiple Trace Modes at the same time?
or only one mode can be applied simultaneously..

IainCRobertson, 2024-07-23:
Yes, they are orthogonal and can be enabled in any combination.

Iain

zhangdujiao, 2024-07-24:
> Yes, they are orthogonal and can be enabled in any combination.

But the ioption field only reports 1 operation mode in one support packet, right?
What if TE in running in two mode together? How ioption can report more operation mode?

IainCRobertson, 2024-07-24:
The ioption field should be a bit vector with 1 bit for each option.  In that way any number can be indicated simultaneously.

Iain

zhangdujiao, 2024-07-25:
thanks, there are 7 modes in Chapter 3.2 totally. The "delta address mode" and "full address mode" are mutually exclusive. Do we need to use 1 bit to represent these two modes? like, 0: delta mode, 1: full mode.
For the implicit return mode, there are there application methods totally: simple counting, partial address compare, full address compare. Do we need to list them all in the ioption field? 
Thanks

IainCRobertson, 2024-07-25:
See inline...

Iain

[Iain] Yes, just 1 bit.

[Iain] There is no mention of full vs partial address compare in the E-Trace spec, so I'm not sure what you're referring to here.  However, I'd expect only a single bit for implicit return mode.  If the hardware has implemented a return stack then I'd expect this to always be used, and the simple counting method to be used only when there is no return stack.  If the hardware has implemented a return stack then the return_stack_size_p discoverable parameter will be non-zero.

closed 2024-07-29 by zhangdujiao