riscv-non-isa/riscv-trace-spec #132: Should each ingress port signal group match separated comparators and filter?

zhangdujiao, 2024-08-01:
If we have two ingress port signal group. Now two ways can be applied:
1.
group0 -> comparator0 -> filter0
group1 -> comparator1 -> filter1

2.
group0 & group1 -> comparator0 -> filter0

which way is better?

IainCRobertson, 2024-08-01:
Option 2.   For example, suppose you want to trace if the instruction address is between A and B.  You only need one comparator unit and one filter to do this.  It shouldn't matter how the hart to encoder interface is set up - single instruction, single group, multiple single instruction lanes or multiple groups.  The user shouldn't have to program the filtering differently depending on the architecture of the processor.

When using groups, the comparator will need to take into account the lower and upper address represented by each group.  For example, address A may be in the middle of the set of instructions retired in that group in that cycle.  When there are multiple groups (or single retirement lanes) the comparator will need to look at all of them in parallel.  If any of the instructions retired in the cycle match the filtering criteria, the recommendation is to trace all the instructions retired in that cycle.

Iain

  1.

zhangdujiao, 2024-08-13:
Very important example, I need to reconsider the design.
For the comparator arithmetic, =, !=, <=, >=, etc...
For example, when a instruction block is sent to the comparator, we can obtain the 1st address and last address of this block.
If the trTeCompPFunction = 0 ("="), the result is true when:
(cfg_value >= 1st address) & (cfg_value <= last address)
For "!=":
(cfg_value < 1st address) | (cfg_value > last address)
For "<"
last address < cfg_value
For ">"
1st address > cfg_value
...
Is this reasonable?

IainCRobertson, 2024-08-13:
Yes, that's exactly right.

Iain

zhangdujiao, 2024-09-07:
Hi Iain, 
a new problem, we can get the last address of the block from : iaddr+2(iretrie-2^ilastsize), that should be correct. However, when we use the address range (1st_addr and last_addr) to match the cfg_value in comparator, the last instruction could be 16-bits or 32-bits wide, and we only count the 1st byte address of the last inst. 
What about the address of last 1 or 3 bytes? If it should be counted in the address range, the last_address should be change to iaddr+2*iretrie -1.
So, should the address of last byte of the last inst. be counted within the address range?
many thanks!

zhangdujiao, 2024-09-07:
If the last address of inst. should be included in te_inst, like at the <next is unqualified> condition. 
Ss this following formula correct?  
last_addr = iaddr+2(iretrie-2^ilastsize)

IainCRobertson, 2024-09-11:
You should base the comparison on the address of the 1st byte of the instruction.

Iain

IainCRobertson, 2024-09-11:
Yes, assuming ^ represents exponentiation.  In other words 2^0 is 1 and 2^1 is 2.

Iain

zhangdujiao, 2024-09-12:
Thanks Iain,
If the ingress port signal is: itype=1/2, iretire=0, which means exc_only. What does the iaddr mean?
When itype=1, does iaddr mean the address of the inst. which occur an exception?
When itype=2, is iaddr meaningless?

Under this condition, if the comparator is configured by address match, how to match if the iaddr is meaningless?

IainCRobertson, 2024-09-12:
Yes, exactly.  For interrupts, the address filtering should be ignored when iretire=0.

Iain

zhangdujiao, 2024-09-13:
> the address filtering should be ignored when iretire=0.

"ignored" means the address shouldn't match when (itype=2 & iretire=0), right?

IainCRobertson, 2024-09-13:
I mean that the value of a the address should not be considered when deciding the overall match result.   For example, if the filter is configured to match on an address range and also an ecause value, the decision about whether this cycle matches the filtering criteria should be based only on ecause.

If you consider the overall match result to be based on the AND of all the individual enabled match conditions, then mathematically the address is treated as if it does match:

Match is true if address matches AND ecause matches

In this cycle, because the address is invalid you have to treat the address as matching for this expression to produce the correct result

Iain

closed 2024-10-09 by zhangdujiao