riscv-non-isa/riscv-trace-spec #135: Does the tval comparison require itype = 1 or 2?

zhangdujiao, 2024-08-18:
When trTeCompPInput codes 2: tval, should the itype =1/2 be added in the if condition, since tval only valid when trap happens.
thanks!

IainCRobertson, 2024-08-19:
Yes, tval should only be compared when itype is 1 or 2.

Iain

zhangdujiao, 2024-08-19:
I see, so the itype =1 or 2 should also be added during ecause match in the filter

IainCRobertson, 2024-08-19:
Yes, absolutely.

Iain

zhangdujiao, 2024-08-22:
For trTeFilterMatchChoiceEcauseLow, "When trTeFilterMatchEcause field for filter #i is set, match all excepion causes for which the corresponding bit is set. If bit N is 1, then match if the ecause is N."
Does the "all excepion causes" means only exceptions? or both excep. and int. ?

zhangdujiao, 2024-08-22:
Can we set 1 at multiple bit? like 'b001001 for trTeFilterMatchChoiceEcauseLow value

IainCRobertson, 2024-08-22:
Yes you can set multiple bits. That’s why it’s a bit vector. ‘Choice’ in the name indicates it is multiple choice, allowing you to pick any combination you want.

Whether it matches interrupts, exceptions or both depends on whether you enable matching on the ‘int’ bit. If this is not enabled, it will match both, and if it is enabled it will match interrupts if the ‘value’ bit is 1, or exceptions if it’s 0.

Apologies for not using the exact field names but I’m on vacation without access to the spec.

zhangdujiao, 2024-08-23:
> Yes you can set multiple bits. That’s why it’s a bit vector. ‘Choice’ in the name indicates it is multiple choice, allowing you to pick any combination you want.

great! have a nice trip:)

closed 2024-08-23 by zhangdujiao