riscv-non-isa/riscv-trace-spec #138: Small suggested clarification

rsnikhil, 2024-09-02:
The word "E-Trace", used on pages 9, 10 and 79, presumably stands for "Efficient Trace", but AFAIK is nowhere so defined.

Suggest adding an item in Section 1.1 Terminology for this.

rsnikhil, 2024-09-02:
‘ucause’ is mentioned in Section 3.2.3; Section 4.1 second bullet; table 4.
‘utval’ is mentioned in Section 4.1 second bullet; Table 4; Table 17; 7.3.2.

AFAIK these CSRs were removed from the Privileged ISA spec?

rsnikhil, 2024-09-03:
In Section 4.1, we have:

    The following information is mandatory:
    ...
    * The instruction_type of retired instructions for:
      * Jumps with a target that cannot be inferred from the source code;
    ...

and

    The following information is optional:
    ...
    * The instruction_type of instructions for:
      ...
      * Jumps with a target that cannot be inferred from the source code;

This seems contradictory (both mandatory and optional)?

IainCRobertson, 2024-09-04:
Rishiyur,
                That's a good observation, and does indeed appear contradictory.  It came about as a result of an earlier clarification attempt.  The list of optional types in section 4.1 corresponds to the more detailed classifications in section 4.11.  Originally, this was broken down into Calls, Tail-calls, Returns, Co-routine swap and Other.  However, there is no formal definition of tail-call in the RISC-V ISA, and so references to 'tail call' in both section 4.1.1 and 4.1 were replaced with 'jump'.  With hindsight, this is too general, and probably needs to be something like "jump with discarded link address" or "call with discarded link address"

Please open an issue against the spec (https://github.com/riscv-non-isa/riscv-trace-spec/issues) and I'll address this in the next update.

Iain

IainCRobertson, 2024-09-04:
> Links:
> You receive all messages sent to this group.
> View/Reply Online (#598)
> To Group
> | Reply To Sender
> | Mute This Topic <https://lists.riscv.org/mt/108261477/6477655> | New
> Topic <https://lists.riscv.org/g/sig-debug-trace-perf-mon/post>
> Your Subscription
> <https://lists.riscv.org/g/sig-debug-trace-perf-mon/editsub/6477655> | Contact
> Group Owner ***@***.***> | Unsubscribe

FWIW, when CTR went through ARC review, ARC requested the following
definitions for the various control transfer types:

[image: image.png]

IainCRobertson, 2024-09-04:
Beeman,
                Thanks – that looks like the terminology we should adopt.  Unfortunately your screen-grab doesn’t show up in the GIT issue – can you paste plain text or include a reference that I can follow later please?

Iain

View/Reply Online (#598)<https://lists.riscv.org/g/sig-debug-trace-perf-mon/message/598> | Reply To ***@***.***?subject=Re:%20Re%3A%20%5BRISC-V%5D%20%5Bsig-debug-trace-perf-mon%5D%20%5Briscv-non-isa%2Friscv-trace-spec%5D%20Small%20suggested%20clarifications%20%28Issue%20%23138%29> | Reply To ***@***.***?subject=Private:%20Re:%20Re%3A%20%5BRISC-V%5D%20%5Bsig-debug-trace-perf-mon%5D%20%5Briscv-non-isa%2Friscv-trace-spec%5D%20Small%20suggested%20clarifications%20%28Issue%20%23138%29> | Mute This Topic<https://lists.riscv.org/mt/108261477/6477655> | New Topic<https://lists.riscv.org/g/sig-debug-trace-perf-mon/post>
Your Subscription<https://lists.riscv.org/g/sig-debug-trace-perf-mon/editsub/6477655> | Contact Group ***@***.***> | Unsubscribe<https://lists.riscv.org/g/sig-debug-trace-perf-mon/unsub> ***@***.******@***.***>]
_._,_._,_

IainCRobertson, 2024-09-04:
Sure, I'll paste the adoc text.

IainCRobertson, 2024-09-04:
It was also improved in N-Trace.  See this:

https://github.com/riscv-non-isa/tg-nexus-trace/blob/main/docs/RISC-V-N-Trace.adoc#2-trace-ingress-port

IMPORTANT: I am against that PR mentioned below.

We agreed to EXTRACT this into SINGLE source (as fast-track).

Thanks,
Robert

IainCRobertson, 2024-09-04:
I see the issue hasn't been filed yet, so pasting the text here:

|INDCALLINH |Inhibit recording of indirect calls.  See <<Transfer Type
Filtering>>.

|DIRCALLINH |Inhibit recording of direct calls.  See <<Transfer Type
Filtering>>.

|INDJMPINH |Inhibit recording of indirect jumps (without linkage).  See
<<Transfer Type Filtering>>.

|DIRJMPINH |Inhibit recording of direct jumps (without linkage).  See
<<Transfer Type Filtering>>.

|CORSWAPINH |Inhibit recording of co-routine swaps.  See <<Transfer Type
Filtering>>.

|RETINH |Inhibit recording of function returns.  See <<Transfer Type
Filtering>>.

|INDLJMPINH |Inhibit recording of other indirect jumps (with linkage).  See
<<Transfer Type Filtering>>.

|DIRLJMPINH |Inhibit recording of other direct jumps (with linkage).  See
<<Transfer Type Filtering>>.

IainCRobertson, 2024-09-04:
Robert,
             The issue is for a request to clarify chapter 4.  Whether that is done as part of an incremental update to the E-Trace spec (along with some other clarifications and corrections) before we pull out chapter 4, or whether chapter 4 gets pulled out first and the clarifications are handled as part of that makes no difference – the important thing is that we have a record of the need for clarification.

Iain

https://github.com/riscv-non-isa/tg-nexus-trace/blob/main/docs/RISC-V-N-Trace.adoc#2-trace-ingress-port

IainCRobertson, 2024-09-04:
N-Trace has 'correct' (ARC-blessed ...) names.
We (me, Jay, Ved=ARC) spent a lot of time (several meetings with ARC) making 'itype' names correctly named and correctly explained.

I assume 'extracted spec' will inherit this.

Thanks,
Robert

https://github.com/riscv-non-isa/tg-nexus-trace/blob/main/docs/RISC-V-N-Trace.adoc#2-trace-ingress-port

IainCRobertson, 2024-09-04:
Robert,
              Yes, and those are the same names Beeman put forward.  Again, no reason not to update the E-Trace spec if it happens that an update is done before chapter 4 is pulled out.

Iain

https://github.com/riscv-non-isa/tg-nexus-trace/blob/main/docs/RISC-V-N-Trace.adoc#2-trace-ingress-port

IainCRobertson, 2024-10-24:
@rsnikhil please review https://github.com/riscv-non-isa/riscv-trace-spec/pull/161 - does this address all of your concerns?

rsnikhil, 2024-11-12:
Yes, all these changes look fine to me (my apologies for my delay in reviewing).

closed 2025-06-02 by IainCRobertson