riscv-non-isa/riscv-trace-spec #142: Clarify/resolve contradictory sentences in Sec 4.1

rsnikhil, 2024-09-04:
In Section 4.1, we have:

    The following information is mandatory:

        * The instruction_type of retired instructions for:
             * Jumps with a target that cannot be inferred from the source code;

and

    The following information is optional:

        * The instruction_type of instructions for:
        ...
           * Jumps with a target that cannot be inferred from the source code;

These seem contradictory (both mandatory and optional)

IainCRobertson, 2024-10-24:
@rsnikhil please review https://github.com/riscv-non-isa/riscv-trace-spec/pull/162 - does this address your concerns?

IainCRobertson, 2025-06-02:
Spec updated and reviewed

closed 2025-06-02 by IainCRobertson