riscv-non-isa/riscv-trace-spec #143: Should filter qualified be 0 or 1 when all the match criteria is not enable

zhangdujiao, 2024-09-18:
when
trTeFilterMatchPrivilege=0
trTeFilterMatchEcause=0
trTeFilterMatchInterrupt=0
trTeFilterMatchComp1-3 =0

what the qualified should be?

IainCRobertson, 2024-09-20:
Apologies for the delay, I've been busy with a workshop for a few days.

The overall filter result is the AND of all the individual criteria matches.  For each criteria, the matching condition behaves as follows:

  *   If the criteria is disabled, this criteria cannot affect the overall filter result, so is considered true (always matches)
  *   If the criteria is enabled, the matching condition is true only if the input matches the criteria, and is false otherwise

Therefore if none of the criteria are enabled, the filter result is always true and trace is always qualified.

Iain

zhangdujiao, 2024-09-20:
Many thanks for the reply, very detailed explanation. problem solved! :)

closed 2024-09-20 by zhangdujiao