riscv-non-isa/riscv-trace-spec #144: interrupt field in the sync-trap packet

zhangdujiao, 2024-09-25:
The sync-trap packet has two types: 1) previous ecause and tval, 2) current ecause and tval. 
The tval should be omitted when itype =2 (interrupt). So, the previous itype and current itype, respectively, should be used to indicate whether the tval field is included at the two circumstance above, right?
What about the 1bit interrupt field? Should it also be determined by previous itype and current itype, or it always be current itype?
Many thanks!

IainCRobertson, 2024-09-25:
Yes, use the previous itype/ecause/tval if thaddr = 1 and the current itype/ecause/tval if thaddr = 0. Same applies for the interrupt field.

Iain

zhangdujiao, 2024-09-25:
> the previous itype/ecause/tval if thaddr = 1 and the current itype/ecause/tval if thaddr = 0

according the Figure 2. Instruction delta trace algorithm, there are three sync-trap packets totally, and it seems the ecause and tval is not depends on thaddr. Like the first trap packet and the last one, thaddr both equals 0, and the first use previous ecause/tval, and the last use current ecause/tval.
Besides the ecause and tval, other fields like pri, branch are all determined by current, right?

Thank!

IainCRobertson, 2024-09-25:
Sorry, you are right; please ignore my previous response.  Let me try again.

As the diagram shows, tval, ecause and interrupt are taken from the previous or current instruction depending on how they are coloured.  All other fields are from the current instruction.

The flow diagram is correct but it's not totally obvious how it relates to the different scenarios.  Let me try to summarise:

  *   Basic case: an interrupt or exception occurs after an instruction which is not an updiscon.
     *   Generate sync-trap when the next instruction (1st instruction of the trap handler) retires, using details of that instruction apart from ecause, interrupt and tval, which are from the interrupt/exception (2nd format 3 box from the top)
  *   Special case 1: as above, but there is another interrupt or exception immediately (before any retirements)
     *   Generate sync-trap on the 2nd exception/interrupt with thaddr = 0, the address if an exception, and ecause/interrupt/tval from the previous block (1st box)
     *   Repeat if there are more back to back exceptions/interrupts
     *   Generate sync-trap when the next instruction (1st instruction of the trap handler) retires, using details of that instruction apart from ecause, interrupt and tval, which are from final interrupt/exception (2nd box)
  *   Special case 2: an exception occurs after an updiscon (i.e. when trying to execute the target of the updiscon)
     *   Generate sync-trap directly from exception, reporting address of the exception (with thaddr = 0).  This is so the decoder can know what the target of the updiscon was (4th box)
     *   Repeat if there are more back to back exceptions/interrupts (so the decoder knows the cause (and address for exceptions) for each trap)
     *   Generate sync-start from the next instruction (1st instruction of trap handler) (3rd box)

Iain

zhangdujiao, 2024-09-27:
Thanks Iain :)
Very detailed explanation!

closed 2024-09-27 by zhangdujiao