riscv-non-isa/riscv-trace-spec #149: What is the correct way to arrange trace payload

zhangdujiao, 2024-09-27:
We can see the example from 13.1.1. Code fragment, line 56.
```
format=1, address=80000104, branches=1, branch_map=0, irreport=0, notify=0, updiscon=0,
Reason[prev_updiscon] Payload[05 04 01 00 80 00]

```
The initial packet should be the order as follow:
trace_packet = {irreport, updiscon, notify, address, branch_map, branches, format};
that is, trace_packet = {1'b0, 1'b0, 1'b0, 0x80000104, 1'b0, 5'b1, 2'b01};

FIRST, compress the trace_packet by using "sign-based" compression method. Supposing the compressed packet is:
trace_packet_compressed = {0x80000104, 1'b0, 5'b1, 2'b01};
 irreport, notify, updiscon and some MSB bits of address are compressed.

NEXT, we need to add 0 to the highest bit of the compressed packet until it is byte-aligned, 

THEN, Invert the big-endian and little-endian to obtain the payload. Since the payload from the example above shows: first byte of the packet is listed in the MSB byte of the Payload.
--am I understand right?

FINALLY, from the e-trace encap spec. The output pakcet format should be:
trace_packet_encap = {payload, timestamp, srcID, header};

For example, 
initial format 2 trace_packet = {irreport, updiscon, notify, address, format};
trace_packet ={1'b0, 1'b0, 1'b0, 0x00000040, 2'b10};
trace_packet_compressed = {8'b0100_0000, 2'b10};   //trace_packet_compressed = {0x40, 2'b10}; 
trace_packet_byte_aligned = 16'b0000_0001_0000_0010;      // trace_packet_byte_aligned = 0x0402;
the final payload = 0x0204;
trace_packet_encap= {0x0204, timestamp, srcID, header};

Is this the format that the spec wants to define? We hope that the decoder software can correctly parse it, so we try our best to follow the protocol definition.. Thanks!

________
Another question from e-trace encap, 2.1.5. Packet Length,
"In cases where the number of bits of payload + srcID is not a multiple of 8, some padding bits are required. These must be placed in the most significant bits of the final byte of the packet."
The "most significant bits of the final byte of the packet" means the MSB of trace_packet_encap above? that is, adding some padding bits in the MSB of payload?

Many thanks!

IainCRobertson, 2024-09-27:
You are on the right lines but if you want to support the general case where the srcID length is not a multiple of 8 bits you take this into account before doing the compression, as the 'length' field in the encapsulated packet indicates the length of the payload + any residual srcID bits (#bits(srcID) % 8).

  *   Assemble the payload
  *   Shift payload left by #bits(srcID) % 8
  *   Sign extend to byte boundary
  *   Determine number of 'same' bytes at the top that can be compressed away and adjust the length accordingly.

An alternative scheme which calculates the length directly without having to know what the uncompressed length would be would be:

  *   Assemble the payload
  *   Shift payload left by #bits(srcID)% 8
  *   Sign extend to 31 bytes
  *   Determine byte position of left-most byte where all bits are not the same.  This is the 'length' value for the header byte

(this is what my implementation does)

Iain

zhangdujiao, 2024-09-28:
> initial format 2 trace_packet = {irreport, updiscon, notify, address, format};
> trace_packet ={1'b0, 1'b0, 1'b0, 0x00000040, 2'b10};
> trace_packet_compressed = {8'b0100_0000, 2'b10}; [...]

The point here is which packet type should be used for compressing.
my example here, trace_packet is used for compressing, since irreport, updiscon, notify and MSB of address are "0" at most of the case. After compressing, trace_packet_compressed is sign-extended to byte boundary, and then get the final payload, which is 0x0204.
If we use the payload rather than initial trace_packet to compress, it could not be an efficient way to compress away many "useless" MSB bits, right?

> * Assemble the payload * Shift payload left by #bits(srcID) % 8 * Sign extend to byte boundary * Determine number of 'same' bytes at the top that can be compressed away and adjust the length accordingly.

So, the last compress process should be done before assemble the payload, right?

IainCRobertson, 2024-09-30:
Sorry, I'm not following your concern.  The exact implementation you chose is of course up to you.  But the important thing to bear in mind is that the result of the compression is to produce a packet with a reduced length that can be uncompressed by sign extension to produce the original.  The 'header' byte of the packet includes the length field, which is used to determine the total length of the packet.  If the number of bits of srcID is not a multiple of 8, this length field covers the payload plus any residual srcID bits.  If you do not align the data to take this into account before doing the compression you may end up with a corrupted packet.  Let me give you an example:

As you have noted, if you compress your trace_packet value in isolation you get a trace_packet_byte_aligned value of 0x0102 (not 0x402 - you made a mistake).  This is 2 bytes long.  This is fine if #bits(srcID) is a multiple of 8.  For example, if it is 8 bits then you will get a packet with a total length of 4 bytes with a header of 0x02, a srcID byte and 2 bytes of payload.

However, suppose the srcID is 7 bits long.  In this case the packet would be (in binary):  address: 01000000 format: 10 srcID: sssssss header: 00000010.  You'll note that the 1 in the address is now the MSB of the packet.  When this is sign extended by the decoder, it will think all the upper bits of the original packet were 1s, which is incorrect.   A correctly formed packet in this case must have a length of 3, with an additional zero byte at the end in order to ensure it decompresses properly.

This is achieved by correctly aligning the trace_packet first:  trace-packet << 7 = 0x0000008100.  When compressing this, we see that a payload of 0x008100 will uncompress correctly, but a payload of 0x8100 will not, so the packet length needs to be 3.

My preferred method to identify the bytes that can be compressed away is to divide the (correctly aligned) payload into overlapping groups of 9 bits.  Each group contains all the bits from a byte, plus the MSB of the preceding byte.  In order to compress a byte away, all 9 bits have to be the same.

Iain

zhangdujiao, 2024-10-02:
> This is achieved by correctly aligning the trace_packet first: trace-packet << 7 = 0x0000008100. When compressing this, we see that a payload of 0x008100 will uncompress correctly, but a payload of 0x8100 will not, so the packet length needs to be 3.

Thanks Iain, I get your point here.
some padding bis should be added in the MSB of the final byte of the encap. packed. And the value of padding bits should be the MSB of the payload whatever 0 or 1.

> The exact implementation you chose is of course up to you.

Does this means our packet compress and encapsulating below are reasonable?

> initial format 2 trace_packet = {irreport, updiscon, notify, address, format}; trace_packet ={1'b0, 1'b0, 1'b0, 0x00000040, 2'b10}; trace_packet_compressed = {8'b0100_0000, 2'b10}; [...]

Your method put one byte as the least compression unit. While I didn't  see the spec has this limitation. Any length of bits can be compressed away if this bits sequence comprises the "sign based compress" condition, right?

Many thanks

> My preferred method to identify the bytes that can be compressed away is to divide the (correctly aligned) payload into overlapping groups of 9 bits. Each group contains all the bits from a byte, plus the MSB of the preceding byte. In order to compress a byte away, all 9 bits have to be the same.

IainCRobertson, 2024-10-03:
See inline...

Iain

[Iain] I think you a typo: trace_packet_byte_aligned  = 0x0102.  You haven't been explicit but I assume you have a 1-bit 'type' field, which is how your final payload is 0x204.  This is the correct answer, but per my previous point you need to align the trace_packet such that the byte boundaries are the same as in the final payload before you compute the compression

[Iain] The encapsulation spec defines a single byte with length = 0 as an idle packet.  So you need at least byte of payload.

zhangdujiao, 2024-10-07:
> The exact implementation you chose is of course up to you.

Thanks Iain, got it!

> I think you a typo: trace_packet_byte_aligned = 0x0102

yes, it should be 0x0102!

> The encapsulation spec defines a single byte with length = 0 as an idle packet. So you need at least byte of payload.

I have thought the idle packet/null packet is used for synchronization, which means the decoder can simply find the trace packet header after the specific synchronization packet.
Should the number of compressed bits also be in bytes? 
like,
trace_packet ={1'b0, 1'b0, 1'b0, 0x00000040, 2'b10}; // 37bits length
trace_packet_compressed shoud be:
{12'b000_0100_0000, 2'b10}
not
 {8'b0100_0000, 2'b10}
I guess the later should be right? any length of bits which are the same as MSB can be compressed away, and a "sign" bit is left in the MSB of compressed packet.

IainCRobertson, 2024-10-07:
See inline...

Iain

[Iain] Yes it is, but this also means that you cannot have a "real" packet with a length of 0 (by which I mean a header byte with length of 0, followed by a SrcID and optional timestamp (but no payload as the length is 0), as this is more than 1 byte total length, and the recipient of the trace stream would have no way to distinguish this.

[Iain] I think it's more logic to compress at the bit level rather than byte level.  And of course, the end result needs to be a number of whole bytes, which in this case is 2.  So the payload needs to sign extended to 16 bits before transmission in any case.

-

closed 2024-10-07 by zhangdujiao