riscv-non-isa/riscv-trace-spec #151: Sequentially inferable jump mode

zhangdujiao, 2024-10-12:
If the sequentially inferable jump mode is enabled. In the following scenarios: 
For the current block, itype is 8, 10, 12 or 14 (updiscon) and sijump = 1, and no trace packets need to be sent at the next instruction block under the condition <updiscon previous>, right? even if the the next itype = 1 or 2.

IainCRobertson, 2024-10-14:
Yes, and no.  sijump = 1 has the effect of converting an uninferrable jump (where the target must be reported) into an inferable one (where the target does not need to be reported).  However, if there are other reasons for reporting the jump target then it must still be reported.  You always have to report the last instruction retired before itype = 1 or 2, otherwise the decoder won't know where in the program the interrupt or exception happened.

Iain

For the current block, itype is 8, 10, 12 or 14 (updiscon) and sijump = 1, and no trace packets need to be sent at the next instruction block under the condition , right? even if the the next itype = 1 or 2.

-

zhangdujiao, 2024-10-14:
sorry, I didn't not express clearly. 
For example: 
t0, sijump=1, itype0=updiscon; 
t1, itype1 ; t2, 
itype2 =0; 
In this scenario, regardless any value of itype1, there is no need to issue trace_packet at t1 to report the jump target address of the updiscon. Is this correct?

IainCRobertson, 2024-10-14:
Ah, I think I see where you're going.  Normally when there is an exception at the target of an updiscon it's necessary to report the address of the exception in the sync-trap message using thaddr = 0.  But if sijump was 1 on the instruction retired before the exception, the target is known to the decoder and so you can use a "normal" sync-trap with thaddr = 0 reporting the address of the trap handler.

Iain

zhangdujiao, 2024-10-14:
OK, I get your point, 
If the instruction block following sijump is not an exception but a normal instruction (itype=1), no packet needs to be issued (to report jump target address), right?

zhangdujiao, 2024-10-14:
> "normal" sync-trap with thaddr = 0

how to understand "normal" sync-trap with thaddr=0

IainCRobertson, 2024-10-14:
Sorry, I meant normal with thaddr = 1.  In other words, a sync trap reporting the address of the trap handler.

Iain

zhangdujiao, 2024-10-15:
> t0, sijump=1, itype0=updiscon; t1, itype1 ; t2, itype2 =0;

The "normal" sync-trap will be issued at t2 by the contidion <exception previous> from my understanding.
At t1, the address of the trap handler will be unknown for the TE, and no packet will be issued, right?

IainCRobertson, 2024-10-15:
Yes, that's exactly correct.

Iain

The "normal" sync-trap will be issued at t2 by the contidion from my understanding.

zhangdujiao, 2024-10-15:
got it, thanks!

closed 2024-10-15 by zhangdujiao