riscv-non-isa/riscv-trace-spec #155: Packet selection when prev_updisocn and next is unqualified are satisfied simultaneously

zhangdujiao, 2024-10-21:
If the last qualified inst. block satisfied the <prev_updisocn> condition, and issue the jump target address of previous jump.
But the last address of this qualified inst. block will not be reported by the condition <next_is_unqualified>, right? Since this qualified inst. block has already issued format1/2 packets.
So, is it ok for this situation which the last address of the trace flow is not reported?

IainCRobertson, 2024-10-21:
That is correct.  The following support packet would then set qual_status to ended_ntr to indicate that the final address has already been reported for other reasons (ntr is an abbreviation for "nothing to report").

Iain

zhangdujiao, 2024-10-22:
That's brilliant and logically coherent! 
Thanks!

zhangdujiao, 2024-11-18:
I am a little confused here, Iain.

If current block (multi-retired inst.) satisfies “prev_updiscon” and "next_is_exception_only" conditions, then two packet will be issued to report the jump target address and inst. addr before trap.
If current block (multi-retired inst.) satisfies “prev_updiscon” and "next_is_unqualified" conditions, then only the jump target address will be reported, and the support-packet (qual_status = 2'b11 (ntr)) will be issued following.

Am I understand right? Why make such a difference on two packets issuing and only one packet issuing for these two circumstances?

IainCRobertson, 2024-11-18:
Dujiao,
             No, there is no difference between these.  For a multi-instruction retirement block:

  *   If the first instruction in the block satisfies "prev_updiscon" then issue a packet reporting the address of the 1st instruction in the block (and any unreported branches from previous blocks - but not the branch status of this block if itype = 4 or 5)
  *   If the last instruction in the block satisfies "next_is_exception_only" or "next_is_unqualified" or next is a sync packet for any other reason - e.g. resync or privilege change - and there are unreported branches, or this block has iretires > 0 and itype  1 or 2, then issue a packet reporting the address of the last instruction in this block (and any unreported branches from this and previous blocks)

If the block is a single instruction then the two conditions are conflated and a single packet is output reporting the address of the only instruction in this block (and any unreported branches from this and previous blocks).

Iain

closed 2024-11-18 by zhangdujiao