riscv-non-isa/riscv-trace-spec #166: ldata/sdata for atomic and CSR data tracing

jaydipdmehta, 2024-10-24:
Etrace spec has below text:

> Atomic and CSR accesses have either both load and store data, or store data and an operand. For CSRs
> and unified atomics, both values are reported via data, with the store data in the LSBs and the load
> data or operand in the MSBs

Its unclear what is carried on 'ldata' and 'sdata' of data trace interface, as above text mentions about LSBs and MSBs. Can such information be detailed?

Meanwhile, can you clarify if below sentences are true?
- For CSRs and unified atomic, 'sdata' carries data thats getting written to address 'daddr' and ldata carries data thats originally read from address 'daddr'?
- Also, for split atomic, same as above query ('sdata' carry data thats getting written to address 'daddr' and 'ldata' carry data thats originally read from address 'daddr'?)
-  For csrrc and csrrs, data being written to csr is reported or rs1 content are expected to be reported on data trace interface?

Thanks,
Jaydip

closed 2025-06-02 by IainCRobertson

j0106223, 2025-09-24:
Hi Iain,

I can’t find the update. Could you please clarify what was updated and which section has the details?

Best regards,
Jack

IainCRobertson, 2025-09-26:
Hmmm I'm confused by that comment too as I didn't make any updates related to atomics in the E-trace spec.  Perhaps I mistakenly updated the wrong issue?

Anyway, as noted elsewhere, I will address this as part of the RHTI separation and will be sure to get you to review the rewording.

j0106223, 2025-09-26:
Hi Iain,

Could you help cross-link the RHTI issue here, so others can find the fix, and reopen this issue until the spec is fixed.

Thanks a lot!

Best regards,
Jack