riscv-non-isa/riscv-trace-spec #167: Should itype=1 or 2 only exists in group0?

zhangdujiao, 2024-10-26:
From the spec, 4.2  Instruction Trace Interface
“The interface supports no more than one privilege change, context change, exception or interrupt per cycle and so signals in groups M and O are not replicated. Furthermore, itype can only take the value 1 or 2 in one of the signal groups, and this must be the newest valid group (i.e. iretire and itype must be zero for higher numbered groups).”

Is that mean itype=1 or 2 only exist in one group. It's not possible for TE to receive two groups of exceptions or interrupts.

IainCRobertson, 2024-10-28:
Yes, itype can be 1 or 2 in only one group.  There cannot be more than one interrupt or exception reported per cycle.  There is only one set of ecause, tval, privilege and context signals.  If there is more than one active group, the interrupt or exception must be the last active.  So you could have the following

Block 0: itype != 1 and itype != 2, iretires > 0
Block 1: itype == 1 or itype = 2, iretires >= 0
Block 2: itype = 0, iretires = 0
Block 3: itype = 0, iretires = 0

But you can't do this:

Block 0: itype == 1 or itype = 2, iretires >= 0
Block 1: itype != 1 and itype != 2, iretires > 0
Block 2: itype = 0, iretires = 0
Block 3: itype = 0, iretires = 0

For the second case, the instructions retired in block 1 for the second case would need typically need a different privilege to block 0.  So any retirements following the interrupt or exception would need to be reported in the next cycle.  In practical terms I would not expect any CPU to want to do this anyway.

Iain

zhangdujiao, 2024-10-29:
got it! thanks!

closed 2024-10-29 by zhangdujiao