riscv-non-isa/riscv-trace-spec #168: Roles of resync counter updating in different trTeControl.trTeInstSyncMode

zhangdujiao, 2024-10-28:
There are three SyncModes totally
1: Count trace messages/packets
2: Count hart clock cycles
3: Count instruction 16-bit half-words
For the 1st "Count trace messages/packets",  whether the update time point is before or after packet sending?
For the "Count instruction 16-bit half-words", whether to update the counter only when the inst. block is qualified? or update the counter, no matter whether it is qualified?
For the "Count hart clock cycles", whether the counter update is only required when trace on (trteactive && trteenable && trtetracing && qualified) or there is no precondition?

Many thanks!

IainCRobertson, 2024-10-28:
See inline...

Iain

[Iain] After.  This is why the algorithm flow chart shows format 0/1/2 sent when the counter equals the threshold, and the sync sent one message later when it exceeds the threshold.

[Iain] This should update with all the half-words retired in a cycle.  So for example, if you have an encoder with 2 blocks you should sum iretires for the 2 blocks - this will be the total number of half-words retired in that cycle.

[Iain] Yes.  No point sending sync messages when trace is off.  Same is true for counting half-words.

zhangdujiao, 2024-10-29:
> This should update with all the half-words retired in a cycle. So for example, if you have an encoder with 2 blocks you should sum iretires for the 2 blocks - this will be the total number of half-words retired in that cycle.

We should count all these two retired isnt. blks, no matter whether it is qualified, right?

IainCRobertson, 2024-10-29:
You should only count instructions that are traced.  If any instructions retiring in a cycle qualify, all the instructions retired in that cycle will be traced.  The whole point of the resync mechanism is to ensure Sync packets are generated at regular intervals in the trace packet stream.

Iain

zhangdujiao, 2024-10-29:
> The whole point of the resync mechanism is to ensure Sync packets are generated at regular intervals in the trace packet stream.

OK I get it!
Also, for "Count hart clock cycles" mode, only the clock cycles under the trace_on condition (qualified==1) are counted, right?

closed 2024-10-29 by zhangdujiao