riscv-non-isa/riscv-trace-spec #172: <Updiscon previous> and <Next inst is exc_only> are satisfied in one block

zhangdujiao, 2024-11-15:
For this situation:
block1 itype = 8 (Uninferable call) ,
block2 itype = 0, iretire >0,
block3 itype = 1, iretire = 0,

Should block2 issue two packets, **te_inst=fmt2 (Updiscon previous)**, and **te_inst = fmt2 (Next inst is exc_only)**.
Or just one packet: te_inst=fmt2 (Updiscon previous), witch the updiscon field in fmt2 should be different from notify to indicate the packet is reporting the instruction following an updiscon and is also the instruction before an exception.

IainCRobertson, 2024-11-15:
Dujiao,
             It depends on the number of instructions in block 2.  If block 2 contains more than 1 instruction then it will generate 2 packets (one containing the 1st instruction in the block, and one containing the address of the last).  But if block2 is just a single instruction then it only needs to generate a single packet.

Iain

zhangdujiao, 2024-11-15:
thx Iain :)

closed 2024-11-15 by zhangdujiao