riscv-non-isa/riscv-trace-spec #176: How to deal with ecause match when exception nesting happens

zhangdujiao, 2024-11-23:
The ecause_match should be asserted when ecause from ingress port equal trTeFilterMatchChoiceEcause, and deasserted when  itype=3 (exception return), right?
For the exception nesting:
t1:  itype=1,  //1st exception, ecause_match=1
t2:  itype=1,  //2nd exception
t3:  itype=3,  //2nd exception return
t4:  itype=3,  //1st exception return, ecause_match=0

The ecause_match should cover the inst. flow from t1~t4, but how to distinguish the exception return from 2nd exception and 1st exception? The ecause_match may be deasserted at t3 instead.

IainCRobertson, 2024-11-25:
Dujiao,
             Yes that's exactly right.  For nested exceptions, my recommendation is to implement a counter, incremented by itype 1 or 2 and decremented by itype 3.  With this you can keep track of how deeply nested the exceptions are, and ensure you don't turn the match off until you return from the outermost exception.

Iain

zhangdujiao, 2024-11-25:
> For nested exceptions, my recommendation is to implement a counter, incremented by itype 1 or 2 and decremented by itype 3

Brilliant, thanks Iain.

closed 2024-11-25 by zhangdujiao