riscv-non-isa/riscv-trace-spec #180: Atomic split for tracing

jaydipdmehta, 2024-11-26:
As per etrace specification, CSR do not have split packet format however atomic has split packet format defined (Split atomic with operand only and Split atomic load data only formats).
When performing read-modify-write for atomic instruction, on trace interface load data can appear along with store data but load data may never be 'split' from its req.

So, its unclear if encoder should use split packet format or unified ones for atomic? (In other words, why split packet formats are needed for atomic when on etrace interface l/sdata (or operand) is presented together?)

Thanks,
Jaydip

IainCRobertson, 2024-11-26:
Jaydip,
             ldata/sdata need not always be presented together.  Table 32 is titled "Packet format for Unified atomic with address and data, but this format also covers split atomics with address and store data only.  The format field value 110 is described as "Unified atomic or split atomic address".  For the latter case, the data field will include store data only.

I will correct this for the next spec version - leave the issue open to act as a reminder for me.

Which format to use will depend on the behaviour of the attached hart.  If you're building a general purpose encoder designed to work with any hart then it should support either, selectable via a top level parameter.

Iain

IainCRobertson, 2025-01-30:
Discussion resolved.  No further action.