riscv-non-isa/riscv-trace-spec #182: Delta addr calculation when last packet emitted by trace encoder was format 0 subformat 1 packet

Muhammad-Saadi-Abbasi, 2024-12-22:
Delta address = current instruction address (that needs to be reported) - last reported address

Consider the following example:
The trace encoder emits the following packets:
- Packet 2 reporting address X
- Packet 0 subformat 0 reporting address Y using jump target cache index
- Packet format 1 wants to report address Z

Packet format 1 wants to report address Z using delta address, how will the delta address be calculated:
Z - X OR Z - Y?

(where X, Y, and Z are full addresses)

According to the trace spec:
```
In delta address mode, addresses are encoded as the difference between the actual address of the
current instruction and the actual address of the instruction reported in the previous packet that
contained an address. 
```
The packets that do not report address are:
- format 3 subformat 2
- format 3 subformat 3
- format 1 (no address, branch map)
- format 0 subformat 0 (no address, branch count)

Based on the answer to the above question, the list would also include format 0 subformat 1 packets.

IainCRobertson, 2024-12-23:
Muhammad,
                        It will be Z - Y.

Iain

Muhammad-Saadi-Abbasi, 2024-12-23:
OK, that makes sense.

closed 2024-12-23 by Muhammad-Saadi-Abbasi

pdonahue-ventana, 2024-12-23:
Could a clarification be added into the current batch of PRs?  It's not perfectly clear that "the previous packet that contained an address" includes packets that include an "index" field instead of an explicitly named "address" field.

IainCRobertson, 2025-01-06:
Yes, I will do that