riscv-non-isa/riscv-trace-spec #185: Why the sync-start packet will not be issued when the first traced instruction is the first in a trap handler

zhangdujiao, 2024-12-31:
sync-start packet contains all the information the decoder needs to fully identify an instruction. It is sent for the first traced instruction (**unless that instruction also happens to be the first in a trap handler**)...
The first instruction of trap handler will not be trace under this circumstance, why is that?

IainCRobertson, 2025-01-06:
Dujiao,
             Sorry, that wording can clearly be improved.  The intent is that the 1st traced instruction will be reported using a sync-start unless that instruction is also the 1st in a trap handler, in which case it will be reported using a sync-trap.

Iain

zhangdujiao, 2025-01-07:
> The intent is that the 1st traced instruction will be reported using a sync-start unless that instruction is also the 1st in a trap handler, in which case it will be reported using a sync-trap.

Thanks Iain, that make sense. 
So, even if previous exception is unqualified, the current qualified block should also issue the sync-trap by the <exception previous> condition.

IainCRobertson, 2025-01-07:
Dujiao,
             The spec isn't specific on that point, but my recommendation would be to issue a sync-start if the exception itself doesn't qualify.

Iain

So, even if previous exception is unqualified, the current qualified block should also issue the sync-trap by the condition.

-

zhangdujiao, 2025-01-07:
> sync-start if the exception itself doesn't qualify

block 0: itype=1, unqualified;
block 1: itype=0, qualified (trace on);
For this situation, is issuing the sync-start the better way rather than sync-trap? 

> The intent is that the 1st traced instruction will be reported using a sync-start unless that instruction is also the 1st in a trap handler, in which case it will be reported using a sync-trap

I find it is difficult to define the 1st traced instruction. For example, block 0 unqualified and block 1 qualified, should block1 be regarded as 1st traced instruction?

IainCRobertson, 2025-01-07:
Dujiao,
            Yes, block 1 is the 1st traced instruction and should be reported using sync-start.

Trace could still start with a sync-trap, but it's a pretty obscure case:

Block 0: unqualified
Block 1: itype = 1 or 2, qualified
Block 2: itype = 1 or 2, qualified

In this case trace will start from sync-trap with thaddr = 0 to report the details of the address of block 1 with the case/tval of block 2.

Iain

zhangdujiao, 2025-01-08:
> Block 0: unqualified Block 1: itype = 1 or 2, qualified Block 2: itype = 1 or 2, qualified In this case trace will start from sync-trap with thaddr = 0 to report the details of the address of block 1 with the case/tval of block 2.

In this case, block 1 will issue sync-start (1st qualified), block 2 will issue sync-trap with thaddr = 0, right?
I cannot see any cases that start from sync-trap if the unqualified exception_previous is not included in the trace algorithm.

IainCRobertson, 2025-01-08:
Dujiao,
             No, I disagree - you can't issue a sync-start for an exception as the exception address doesn't retire.  But there was a silly mistake in my last response.  The 1st trace packet should be a sync-trap with thaddr = 0 reporting the address of block 2 with cause/tval of block 1.

Iain

zhangdujiao, 2025-01-08:
Thanks Iain, I get your point here.

AoteJin, 2025-02-10:
Hi @zhangdujiao @IainCRobertson ,

Block 0: unqualified 
Block 1: itype = 1 or 2, qualified 
Block 2: itype = 1 or 2, qualified 

Regarding the sequence, I am thinking about the generalized case where Block 1 could also be a normal instruction generating trap and Block 2 is at the top of trap handler. In this case, if the 1st trace packet is {thaddr=0, addr=Block2, tval/cause=Block1}, we lose the information where the first exception is generated, that is address of Block 1. What decoder can reconstruct is that somewhere in code is erroneous (by tval/casue=Block 1) and 1st instruction in trap handler throws exception (by thaddr=0, addr=Block2). 

My point is that when Block 1 is qualified and throws exception, the user is likely to be interested in its address to identify the potential issue. Given the fact that trace can be enabled by trigger-on with NAPOT encoding, the first qualified instruction may not be easily deduced otherwise.

IainCRobertson, 2025-02-11:
Aote,
          That's a fair point.  This could be addressed by treating an exception on the 1st qualifying instruction in the same way that an exception at the target of an uninferable discontinuity is treated (see section 7.3.1).

That said, I think changing the spec to require that would constitute a normative change, which would require the spec to be re-ratified.  If we added a note that this was recommended behaviour for this corner case that may be acceptable.  @***@***.******@***.***> what do you think?

Iain

yandumchin, 2025-02-16:
Hi @IainCRobertson ,
Thanks for clarification on this corner. Another corner occurs when last qualified instruction (before trace lost or disable) is itype=1 or 2. In this case, is it expected to send trap packet {thaddr=0, addr=last qualified instruction, tval/cause=last qualified instruction}?

zhangdujiao, 2025-02-17:
> Hi [@IainCRobertson](https://github.com/IainCRobertson) , Thanks for clarification on this corner. Another corner occurs when last qualified instruction (before trace lost or disable) is itype=1 or 2. In this case, is it expected to send trap packet {thaddr=0, addr=last qualified instruction, tval/cause=last qualified instruction} prior to support packet?

The fundamental principle when an exception occurs is to report both the qualified instruction address before the exception and the qualified instruction address after the exception (trap handler address). 
If the exception is triggered by the last qualified instruction, I think there is no need to send sync-trap with thaddr=0. This is because the pre-exception instruction address has already been reported (through formats 0/1/2, where the updiscon field indicates whether the address corresponds to the pre-exception instruction). By parsing formats 0/1/2 via the decoder, the address where the exception occurred can be determined.

IainCRobertson, 2025-02-17:
Re-opening as Aote's point is not yet fully addressed.

yandumchin, 2025-02-17:
Thanks @zhangdujiao

IainCRobertson, 2025-02-17:
I don't think there is anything wrong with that definition, and nor do I think there is any need for a "special meaning".  The final retired instruction before the trap is reported via the format 0/1/2.  If it also happens to be the target of an updiscon, the 'updiscon' bit will be set, and this ensures that the decoder always follows the execution path to the right place for the reasons explained in the spec.   If the instruction which traps happens not to be traced because of a trace-off trigger or because the filtering criteria is no longer met, the final retired instruction before trace ends has still been correctly reported.

AoteJin, 2025-02-18:
> That's a fair point.  This could be addressed by treating an exception on the 1st qualifying instruction in the same way that an exception at the target of an uninferable discontinuity is treated (see section 7.3.1).

I agree it is a valid solution, and a note should be good enough. Based on this, I assume when both the 1st qualified instruction and trap handler entry throw exceptions, there should be back-to-back trap packets with thaddr=0, isn't it?

E.g.:
Block 0: unqualified
Block 1 (1st qualified inst) : itype = 1 or 2, qualified    -> trap packet {thaddr=0, addr=Block1, tval/cause=Block1}
Block 2(trap handler entry): itype = 1 or 2, qualified    -> trap packet {thaddr=0, addr=Block2, tval/cause=Block2}

AoteJin, 2025-02-19:
Hi @zhangdujiao 

>(through formats 0/1/2, where the updiscon field indicates whether the address corresponds to the pre-exception instruction). By parsing formats 0/1/2 via the decoder, the address where the exception occurred can be determined.

In @yandumchin 's example where last qualified instruction throws exception, I agree that last retired instruction will be reported by format 1/2 without generating sync-trap. But the `updiscon` field in packet is not to indicate the trap happened after reported instruction, but only to notify decoder whether the address is a target of a previous updiscon.

In fact, I don't think the decoder should even be aware of the trap, since all information related to trap belongs to unqualified information.

AoteJin, 2025-02-26:
I have no more concern regarding this issue. @zhangdujiao  could you help close it. Thanks!

closed 2025-06-02 by IainCRobertson