riscv-non-isa/riscv-trace-spec #186: The address of the instruction which caused the 2nd trap should be reported

Muhammad-Saadi-Abbasi, 2025-01-03:
When there are back-back exceptions (1st exceptions is not on the an uninferable PC discontinuity) then the trace encoder emits the following packets (as per https://github.com/riscv-non-isa/riscv-trace-spec/issues/126):
- Format 1 or 2 indicating the last instruction retired before the trap
- Sync-trap with thaddr = 0 reports the cause of the 1st trap (and the address of the instruction which raised the exception).
- Sync-trap with thaddr = 1 reports the cause of the 2nd trap and the address of the 1st instruction that retires after the 2nd trap

Consider the following scenario:

1st instruction traps (due to load access fault), this trap is taken in S mode (because it is delegated), and then the 1st instruction of the supervisor trap handler also traps (due to something that is not delegated to S mode so it is taken in M mode). Then the 1st instruction of the machine trap handler retires. 

Now combining both the information:
- Format 1/2 reports the address of the instruction retired before the 1st trap
- Format 3.1 reports the address of the instruction that traps (1st trap)
- Format 3.1 reports the address of 1st instruction of the M mode trap handler

This is not a real-world scenario but how will the end user know the address of the instruction that caused the 2nd trap?

IainCRobertson, 2025-01-06:
Muhammad,
                       The text you quote from #126 is a little ambiguous.  It should read (clarifications in green):

  *   Format 1 or 2 indicating last instruction retired before the 1st trap
  *   Sync-trap with thaddr = 0 reports the cause of the 1st trap (and the address of the instruction which raised the 2nd trap)
  *   Sync-trap with thaddr = 1 reports the cause of the 2nd trap and the address of the 1st instruction that retires after the 2nd trap

Iain

Muhammad-Saadi-Abbasi, 2025-01-07:
Thanks, it makes perfect sense.

closed 2025-01-07 by Muhammad-Saadi-Abbasi