riscv-non-isa/riscv-trace-spec #190: Clarify When sdata is Valid

j0106223, 2025-01-17:
On page 26, Table 14 (Data Interface Signals), the validity of sdata should depend not only on dretire being high but also on dtype being store.

IainCRobertson, 2025-01-31:
Fixed - please review: https://github.com/riscv-non-isa/riscv-trace-spec/pull/195

j0106223, 2025-02-27:
Thank you all for your help.

j0106223, 2025-09-03:
Hi Iain,

In Section 4.4 Data Trace Interface, it states:

"_Atomic and CSR accesses have either both load and store data, or store data and an operand. For CSRs
and unified atomics, both values are reported via data, with the store data in the LSBs and the load
data or operand in the MSBs_."

My question is: if both the load and store data of an atomic access are already reported via data, why should sdata still be valid when dtype is 8–14?

From the text, it only explains the case for unified atomics, but does not describe how non-unified atomics are reported. Is sdata intended for operations involving non-unified atomics?

Could you also provide an example showing how a hart reports atomics via sdata and ldata?

IainCRobertson, 2025-09-08:
Jack,
         The distinction here is between unified and split transactions.

Stores and CRS are always unified.
Loads and atomics can be either unified or split.

Unified atomics and CSRs (which are always unified) use the 'data' bus, with the store data in the lower order bits and the load data or operand in the upper bits.

Split atomics use ldata for the load data or operand and sdata for the store data.

Iain

j0106223, 2025-09-17:
Hi Iain,

For Unified Atomics, is the value from rs2 also reported in the low part of data?

Best regards,
Jack

IainCRobertson, 2025-09-17:
Yes, always.  The current spec uses the word 'operand' and that is what it's referring to.  As I said, I think this could be clearer and I'll endeavour to address that.

Also note that with split atomics (now I've remembered how this is supposed to work) it isn't necessary to wait for the load data to be returned before issuing the dtype and sdata.

Iain

IainCRobertson, 2025-09-26:
> Hi Iain,
>
> I have another question about CSR accesses.
>
> Since dtype shows the CSR access type and the operand is known, the decoder should be able to calculate the CSR value after update, similar to our earlier conclusion for atomic accesses. Which of these cases is reported?
>
> Case 1: data = {rs1/uimm, updated CSR value}
> Case 2: data = {rs1/uimm, original CSR value}

Jack,
         This a (very misleading) typo in the  spec.  It should read

both values are reported via data, with the store data or operand in the LSBs and the load data in the MSBs.

For all atomics, ldata (or MSBs of data) must correspond to the value loaded into rd, and sdata (or LSBs of data) must correspond to the value from rs2.

For CSRs, ldata (or MSBs of data) must correspond to the original value of the CSR written to rd, and sdata (or LSBs of data) must correspond to the value in rs1.

Arguably "store data or operand" should just be "operand".  The actual stored value is always obtained by applying the operator indicated by the instruction (and encoded in dtype) to this value.  In some cases the actual stored value will end up being the same as this operand but that is besides the point.

I will  fix this as part of the ongoing work to split out the interface into the standalone RHTI spec.  I'll tag you when I have a PR ready for review.

Iain

IainCRobertson, 2025-09-26:
See https://github.com/riscv-non-isa/hart-trace-interface/issues/31

IainCRobertson, 2025-09-26:
> Hi Iain,
>
> The content you mentioned seems to have a typo; ldata and sdata should not appear because CSR access is always reported in a unified form.

Good point - I forgot that at the time!

Iain

j0106223, 2025-10-01:
Review completed, no issues. Thanks.

j0106223, 2025-10-23:
Hi Iain,

In this thread, you mentioned that CSRs are always unified. However, in [RISC-V][soc-infra] Internal Review for RISC-V Hart to Trace Interface (RHTI) Specification [#987](https://lists.riscv.org/g/sig-debug-trace-perf-mon/message/987), there is discussion regarding the use of a split interface for CSRs, and it is noted that split and unified implementations are expected to be mutually exclusive.

Reopen this issue until a clearer conclusion is reached.

Best regards,
Jack

IainCRobertson, 2025-10-23:
> Hi Iain,
>
> In PR#33<https://github.com/riscv-non-isa/hart-trace-interface/pull/33> Clarified atomic and CSR access descriptions, the behavior of unified and split atomics has been clarified, but CSRs have not been clarified in the same way.
> Would it be possible to add a description for split CSRs as well?

Jack,
         Yes, I will add this.  Leave the issue open as a reminder for me.

Iain