riscv-non-isa/riscv-trace-spec #199: Clarify or Improve Description of `branches` Field in Packet Format 1

AoteJin, 2025-02-12:
The description for `branches` is messy and unclear.

[image]
[image]

According to the context, I suppose it intends to convey that `branches`=0 cannot occur for layout in Table 21 while in layout of Table 22 `branches` is fixed to 31.  Is my understanding correct?

IainCRobertson, 2025-02-12:
Aote,
          The formatting here got messed up in the conversion from LaTeX to adoc.  This is already fixed in one of the PRs for the next release: https://github.com/riscv-non-isa/riscv-trace-spec/pull/157/files

Iain

Screenshot.2025-02-12.at.13.24.32.png (view on web)<https://github.com/user-attachments/assets/cdd1ddbf-b74a-41cd-9342-566c59b1b598> Screenshot.2025-02-12.at.13.24.10.png (view on web)<https://github.com/user-attachments/assets/b5f5e6d6-f6c4-4139-b01c-049b57792a70>

Screenshot.2025-02-12.at.13.24.32.png (view on web)<https://github.com/user-attachments/assets/cdd1ddbf-b74a-41cd-9342-566c59b1b598> Screenshot.2025-02-12.at.13.24.10.png (view on web)<https://github.com/user-attachments/assets/b5f5e6d6-f6c4-4139-b01c-049b57792a70>

AoteJin, 2025-02-13:
Thanks for heads up! Closing issue.

closed 2025-02-13 by AoteJin