riscv-non-isa/riscv-trace-spec #201: Should a sync packet populate the jump target cache when reporting the target of an uninferable jump?

Muhammad-Saadi-Abbasi, 2025-02-13:
The purpose of the sync packet is synchronization, and it can report any type of instruction. If it reports the target of an uninferable jump, should it populate the jump target cache?

IainCRobertson, 2025-02-13:
Mohammad,
                       In most cases no, but It depends on resync_mode.

If resync_mode == 0, no periodic sync packets will be generated and the assumption here is that the decoder will always have visibility of the sync packet from the start of trace.  In this case, the sync packet at the start of trace should not populate the JTC, because the decoder does not know that it is a jump target - the jump itself was not traced.  Without knowing it's a jump target the decoder cannot update its own implementation of the JTC, and so if the encoder had updated the JTC the encoder and decoder's JTCs would not be coherent.  For subsequent sync packets, it is okay to populate the JTC because the decoder will know about the jump which occurred immediately before the sync, and so knows the sync is reporting a jump target and will update its cache.

If resync_mode > 0, periodic sync packets are generated and the assumption here is that the decoder may start decoding from any sync packet.  In this case, no knowledge about instructions prior to the sync packet can be assumed and so the JTC must not be populated.

Iain

Muhammad-Saadi-Abbasi, 2025-02-13:
> For subsequent sync packets, it is okay to populate the JTC because the decoder will know about the jump which occurred immediately before the sync, and so knows the sync is reporting a jump target and will update its cache.

How does the decoder know that the sync packet is reporting the address of an uninferable jump target because it may be emitted due to trace loss?

Muhammad-Saadi-Abbasi, 2025-02-13:
Oh, OK. When trace data is lost, the encoder first emits a support packet and then a sync packet. This allows the decoder to determine whether the sync packet was emitted due to trace data loss. Thanks, what you said makes sense.

closed 2025-02-13 by Muhammad-Saadi-Abbasi