riscv-non-isa/riscv-trace-spec #204: Clarification required when updiscon field has to be reported and the address also hits the in the JTC

Muhammad-Saadi-Abbasi, 2025-02-19:
Consider the following sequence of instructions:

jalr x0, (x8)
instr x ---> Target of above uninferable jump
ecall

In the above sequence, the address of instruction x has to be reported with updiscon field set (notify != updiscon) and instruction x also gets a hit in the jump target cache. Instr x can't be reported using format 0.1 packet because it doesn't have the updiscon field and in this case, it still has to be reported using format 1/2 or maybe 0.0 if there are more than 31 correct predictions, right?

If you agree with my statement above then can you please add a clarification in the following sentence of the spec:
_However, the encoder may still choose to output the differential address using format 1 or 2 if the resulting packet is shorter_ or if the updiscon field is set.

IainCRobertson, 2025-02-19:
Muhammad,
                       I disagree.  The reason the format 1 / 2 includes the updiscon bit is so the decoder can distinguish the case where an address is reported as the target of an uninferable jump, from the case where the same address is reported as part of a linear sequence because there is a sync packet coming next (see section 7.6.2).  A format 0.1 is only ever used if the address is reached as the target of an uninferable jump, and so there is no potential ambiguity from the decoder's perspective, and no need for the updiscon bit, which is why it is not included in the format 0.1 packet.

Iain

IainCRobertson, 2025-02-19:
But my point is that the updiscon bit doesn't need to be set if you're using a format 0.1 because a format 0.1 will _only_ be used for reporting an updiscon target, so there is no need for an updiscon bit.  If that address were reached via linear fall-through and not as the target of an uninferable jump, there would be no cache hit and a format 1 or 2 would be used.

The target of the uninferable jump would be reported regardless of whether there is a trap at the next instruction so the decoder doesn't need to know that there is a trap next.  The fact that a format 0.1 is used means it must be an updiscon target.  Another way to think about this is that the format 0.1 has an implicit updiscon bit that is always set.

Muhammad-Saadi-Abbasi, 2025-02-19:
I understand what you mean. Thank you very much for the detailed explanation.

closed 2025-02-19 by Muhammad-Saadi-Abbasi