riscv-non-isa/riscv-trace-spec #205: Invalid  iaddr when itype=2

AoteJin, 2025-02-24:
[image]

The table's description suggests that the iaddr is invalid when itype=2. In scenarios where an interrupt occurs immediately after a mret instruction, the previous block is reported when retiring mret, and the interrupt will be reported with {itype=2, iretire=0}. Therefore, it may be more accurate to state "Invalid iaddr when itype=1 or 2".

IainCRobertson, 2025-02-24:
Aote,
          I'm not following your thinking here, but disagree that the address can be invalid when itype = 1.  When itype = 1, the address input needs to report the address of the instruction that caused the exception.  Interrupts occur between instructions and so there is no meaningful address for an interrupt.

Iain

Screenshot.2025-02-24.at.16.28.28.png (view on web)<https://github.com/user-attachments/assets/a15628fb-29c7-46aa-96c6-0d09cd7b4d0f>

Screenshot.2025-02-24.at.16.28.28.png (view on web)<https://github.com/user-attachments/assets/a15628fb-29c7-46aa-96c6-0d09cd7b4d0f>

AoteJin, 2025-02-24:
Hi Iain,

I agree that  "_Interrupts occur between instructions_ " . But let's take the following sequence as an example:

1. An interrupt delegated to S-mode, e.g. S-mode software interrupt,  is pending because trap handler in M-mode is still handling the previous trap.
2. The trap handler executes mret to return to S-mode
3. The pending Interrupt causes a trap immediately after the mret (before any instruction get retired)

I think the transaction appears on ingress port should be: 
1. A transaction after mret with iretire>0 and itype=3 to indicate the mret
2. The following transaction with iretire=0 and itype=2 to show the interrupt

IainCRobertson, 2025-02-24:
Aote,
          I agree, but what has that to do with the validity or otherwise of the value on the address input when itype=2?  The encoder will ignore the address when itype is 2, so it is a don't care.  The interrupt occurs after the mret, and the 1st important address is that of the interrupt's trap handler, which will come from the next retirement block.  The address of the instruction which would have been retired after the mret if the interrupt had not occurred will be reported later, as it will be the instruction retired (or faulted) after the *ret that terminates the interrupt handler.

Iain

AoteJin, 2025-02-25:
Hi Iain,

I got your point. In contrary to the exception, EPC of an interrupt does not contribute to re-constructing the trace and it is sufficient to inform decoder that there is an interrupt occurring after an updiscon. Thus, iaddr can be invalid when iretire=0 and itype=2. Is my understanding correct?

IainCRobertson, 2025-02-25:
Aote,
          Yes, exactly.

Iain