riscv-non-isa/riscv-trace-spec #206: Handling of Return Instructions in JTC with Implicit Return Mode

Muhammad-Saadi-Abbasi, 2025-02-26:
The trace encoder is running with implicit return mode and jump target cache (JTC) mode enabled. If a return instruction matches the head of the Return Address Stack (RAS), the encoder will not report it. However, the question is: will it populate the JTC?

The spec states:

>Each uninferable jump target is first compared with the entry at its index in the cache. If it is found in the cache, the index number is traced rather than the target address.

The definition of uninferable jumps is:

>Inferable jump: A jump where the target address is supplied via a constant embedded within the jump opcode.
>Uninferable jump: A jump that does not meet the above definition.

However, the implicit return mode section states:

>The encoder can maintain a stack of expected return addresses and only treat a return as inferable if the actual return address matches the prediction.

By definition, a return instruction is uninferable. However, if the return address matches the prediction, it is treated as inferable in implicit return mode. 

In my opinion, if a return instruction matches the head of the RAS, it shouldn't populate the JTC, as this would only pollute the cache.

IainCRobertson, 2025-02-26:
Muhammad,
                       I agree with your reasoning, and it's a reasonable thing to infer from the spec, but as currently written the spec does not spell this out.

I will add a clarification.

Iain

IainCRobertson, 2025-02-27:
See https://github.com/riscv-non-isa/riscv-trace-spec/pull/208

IainCRobertson, 2025-04-08:
@pdonahue-ventana regarding the concerns about timing impact, a legitimate mitigation for this case would be to not update the contents of a JTC entry with the address of a mis-predicted return, but instead to just invalidate the entry it would have occupied.  Timing-wise invalidating should be less critical as it doesn't require driving a heavily loaded signal controlling update of all the address bits in the entry, or a mux to select which address to use in the case of multiple returns in the same cycle for a superscalar processor.

Mis-predicted returns are in themselves a very rare occurrence, so the performance impact will be negligible.  Invalidating instead of updating with the actual return target address will only make a difference if there is another indirect call or mis-predicted return to the same address before the cache entry is reused, and this will be an even rarer occurrence!

The encoder already makes the decision on whether to report the address using a format 2 or format 0 JTC packet anyway, and this proposal would just be forcing it to use the former format for this case, so the decoder is not impacted.

I can add an informative recommendation for this - what do you think?

pdonahue-ventana, 2025-04-25:
Sorry for the delayed response but I do like your idea.  Upon my first reading, I didn't understand how you could choose to do something like that without getting the encoder and decoder out of sync.  Now I understand that the encoder can randomly invalidate the JTC at any time.  If it's invalid then nothing can hit, the encoder can never send out a packet with that index, and it doesn't really matter what the decoder thinks is in that index.

It would be helpful to mention that, especially since it wasn't immediately obvious.