riscv-non-isa/riscv-trace-spec #211: Constraint on sequentially inferable jump

AoteJin, 2025-03-05:
The normative text defines the sequentially inferable jump as:

_"Such jump targets are classified as sequentially inferable if the pair of instructions are retired consecutively (i.e. the auipc, lui or c.lui immediately precedes the jump)."_

However, it actually implies that the rd of auipc, lui/c.lui should be equal to the rd of next jump, like what is defined in decoder pseudo code:

>if((prev_instr.opcode == AUIPC) or
        (prev_instr.opcode == LUI) or
         (prev_instr.opcode == C.LUI))
return (instr.rs1 == prev_instr.rd)

Should we also add such constraint on normative text?

IainCRobertson, 2025-03-06:
Aote,
          I don't think anything needs changing here.  The full normative text is

However, there are some jump targets which can still be deduced from the binary executable by
considering pairs of instructions even though by the above definition they are classified as
uninferable. Specifically, jump targets that are supplied via
* an lui or c.lui (a register which contains a constant), or
* an auipc (a register which contains a constant offset from the PC).
Such jump targets are classified as sequentially inferable if the pair of instructions are retired
consecutively (i.e. the auipc, lui or c.lui immediately precedes the jump).

I think the highlighted part makes clear that the rd must be the same, as if they are not then the jump target is not "supplied via" one of the specified instructions in the "pair or instructions".

Iain

pdonahue-ventana, 2025-03-08:
I think that it would be an improvement to say "when the source register for the jump instruction is supplied via" or "...rs1 for the jump instruction is supplied via" rather than "jump targets that are supplied via."  The jump target is actually supplied by a combination of the lui/auipc and the jump instruction itself.  (It would be correct to say "jump targets that are partially supplied via" but that would just be obscuring things.)

IainCRobertson, 2025-03-11:
That works for me.  @AoteJin are you happy with this: https://github.com/riscv-non-isa/riscv-trace-spec/pull/213?

AoteJin, 2025-03-12:
Thanks @pdonahue-ventana @IainCRobertson , I think it makes the idea more precise. Works for me too.

closed 2025-04-24 by IainCRobertson