riscv-non-isa/riscv-trace-spec #218: Priv field in ingress port and definition of "privilege change"

AoteJin, 2025-03-20:
Hi,

the spec defines priv field in ingress port as  :
_"priv[privilege_width_p-1:0]   Privilege level for all instructions retired on this cycle."_

But it appears to be ambiguous when it comes to a block ends with mret/sret where the privilege may change when retiring the last instruction. Moreover, it also might complicate the hart in which the exception/interrupt results in privilege change. That is, when a block is reported by exception/interrupt, the hart can already changed to a new privilege once it occurs. 

Another unclear case is that the synchronization on:
* If the privilege level changes; 

When mret executed, the privilege before equals the one after instruction. Does it count for a privilege level change? At first glimpse, I thought no. But the branch predictor or implicit return counter/cache relies on sync to get cleared. It might make sense to clear the leftover states of trap handler when return to normal execution, since it will improve efficiency.

pdonahue-ventana, 2025-03-21:
> But it appears to be ambiguous when it comes to a block ends with mret/sret. The privilege may change while retiring the last instruction.

My understanding is that mret changes privilege mode immediately after the instruction (just like mret or any branch/jump/return changes the PC immediately after the instruction).  I know that I've seen that described somewhere but it might have been on a mailing list.  For what it's worth, ChatGPT agrees: "In RISC-V, the privilege mode change occurs immediately after the execution of the MRET instruction."  If that's correct then I don't think that there's an ambiguity.

AoteJin, 2025-03-21:
Fair enough. I think people can argue xret belongs to privilege level before the its execution and thus priv field always reports the privilege before execution. This made me think further, in case of exception/interrupt block, there might be no instruction retired within it, and the priv field can be ignored in such cases.

IainCRobertson, 2025-03-21:
Privilege change means just that: a change in privilege.  It is detected when the value of the privilege signals in the hart-trace interface differ between one valid cycle and the next (a valid cycle being retires > 0 or itype = 1 or 2).

AoteJin, 2025-03-21:
Sure, cannot agree more. I am thinking it might be worth to have itype=3 (trap return) also trigger resynchronization. Because the horizontal trap will be handled in same privilege level while execution content of trap handler are thoroughly irrelevant to the normal program. The leftover states (e.g., in branch predictor) of trap handler harms the compression rate.

IainCRobertson, 2025-03-21:
Well that would be a normative spec change, and besides, I disagree that "leftover" state harms the compression.  The start state for the branch predictor after synchronisation is not inherently better or worse for the code that comes after the trap handler.

AoteJin, 2025-04-09:
"The start state for the branch predictor after synchronisation is not inherently better or worse for the code that comes after the trap handler."

That's true, hard to tell whether resynchronization can help or not. Thanks for clarification and closing the issue.

closed 2025-04-09 by AoteJin