riscv-non-isa/riscv-trace-spec #219: The notify field in format0/1/2

zhangdujiao, 2025-03-21:
"If the value of this bit is different from the MSB of address, it indicates that this packet is reporting an instruction that is not the target of an uninferable discontinuity because a notification was requested via trigger[2]"
What if the packet is issued by trigger[2], and the address is also the target of an updiscon at the same time.
In this case, what should the notify field be?

IainCRobertson, 2025-03-21:
As the description says, it's used for an instruction that is not the target of an uninferable discontinuity.  So for all iferrable discontinuity targets it should be the same value as MSB of address regardless of whether there was a trigger. 

I would argue that the packet was not issued by the trigger as it would always be issued anyway because it's an uninferable discontinuity target.

The purpose of this bit is to ensure that the decoder operates correctly when an address which is not an uninferrable discontinuity target is issued.   The corner case is described in the E-Trace spec: an address could be reached either sequentially or as a jump target and the decoder needs help to know which case it is in order to correctly follow the execution path.

zhangdujiao, 2025-03-21:
Thanks Iain, well explained.

closed 2025-03-21 by zhangdujiao