riscv-non-isa/riscv-trace-spec #223: Branch count value when reported address is for branch

AoteJin, 2025-04-08:
Hi, 

I read through the below text and try to figure out the branch_count value  (format 0, subformat 1) when the reported address is for a branch.

> When a branch count is reported without an address it is because a branch has failed the prediction.
> However, when an address is reported along with a branch count, it will be because the packet was
> initiated by an uninferable discontinuity, an exception, or because a branch has been encountered that
> increments branch_count to 0xffff_ffff. For the latter case, the reported address will always be for a
> branch, and in the former cases it may be. If it is a branch, it is necessary to be explicit about whether
> or not the prediction was met or not. If it is met, then the reported address is that of the last correctly
> predicted branch.

It seems indicating that the branch count includes the prediction result of the current branch by the description "because a branch has been encountered that increments branch_count to 0xffff_ffff." My previous understanding was that the branch_count would not increment when the reported address is for a branch and consequently branch_fmt is added to denote the prediction result. That is, the branch brings branch_count to 0xffff_fffff will not generate a packet, but the next branch will. Is my understanding correct?

IainCRobertson, 2025-04-08:
Aote,
          I'm not totally following but it seems like you may be conflating several different conditions here.  There are 4:

  *   Case 1: the target of an updiscon, or the final retired instruction before a trap must be reported.  The reported instruction is not a branch.
     *   This will use branch_fmt = 10
     *   branch_count accounts for all branches preceding the reported address (all correctly predicted branches)
  *   Case 2: as case 1 but the reported instruction is also a branch, and the outcome of that branch is correctly predicted
     *   This will also use branch_fmt = 10
     *   branch_count accounts for all branches preceding and including the reported address (all correctly predicted branches)
  *   Case 3: as case 2 but the outcome of the branch is mispredicted
     *   This will use branch_fmt = 11
     *   branch_count accounts for all branches preceding the reported address (all correctly predicted branches)
  *   Case 4: A well predicted branch causes the branch_count to reach the maximum value.  There is no reason to report the address, but the branch count must be reported before it rolls over, so a packet is generated in the same way as case 2, reporting the address of the branch.
     *   This will use branch_format = 10
     *   branch_count accounts for all branches preceding and including the reported address (all correctly predicted branches)

The formatting of the branch_fmt field in Table 23 is also messed up (will be fixed in the upcoming release), and I don't think this helps!

Iain

AoteJin, 2025-04-09:
Hi Iain,

Thanks for sum up the scenarios! The point confuses me is that the we can differentiate the case 1/2/3/4 with only branch_count and parsing the instruction at reported address. It seems to me no need to involve addition branch_fmt field in the packet. Did I miss anything here?

IainCRobertson, 2025-04-09:
Aote,
          Are you still confused?  The distinction between the cases is as follows:

  *   For all cases, the decoder behaviour is the same: it follows the execution path, working through the branch count/predictions, until it reaches the reported address or an uninferrable jump when all branches have been accounted for.  If the reported address is encountered without encountering an indirect jump, decoding stops and waits for the next packet (unless the updiscon bit is different from notify).   This is fundamentally the same as processing a format 1 packet except it's using a count of predicted branches rather than a map of taken/not taken flags.
     *   Case 2 and 4 look identical to the decoder
     *   For cases 2, 3, and 4 branch_fmt determines where the decoder goes after the final branch when decoding the next packet
  *   This is only a distinct case for the encoder - it has to output a packet when the branch count reaches the max otherwise branch history would be lost due to roll over.

Iain

AoteJin, 2025-04-09:
Hi Iain,

This is where I got confused:

>    *  For cases 2, 3, and 4 branch_fmt determines where the decoder goes after the final branch when decoding the next packet

As you said in previous post:

>   *   Case 2: as case 1 but the reported instruction is also a branch, and the outcome of that branch is correctly predicted
         *   This will also use branch_fmt = 10
         *   branch_count accounts for all branches preceding and including the reported address (all correctly predicted branches)

the prediction result of reported address for current branch will be included in branch_count. That appears to be that the decoder decrements branch_count to 1 when reaching the reported address (as the value 1 represents the correct prediction of the current branch).

And the in case 3 the branch_count will be decremented to 0 when decoder reaches the reported address. It seems branch_fmt is not must-have to determine where to go after the reported branch.

IainCRobertson, 2025-04-09:
Aote,
          You cannot simply use whether the branch count is 0 or 1 when the reported address is encountered to determine whether the final branch was correctly or incorrectly predicted, as the reported address may be encountered by the decoder when following the execution path before it reaches the place where it should stop.  Using your suggestion the decoder may stop prematurely.

Iain

AoteJin, 2025-04-09:
Now I see your point. Basically, when decoder follows the execution path till the reported address with the branch_cout=1, it cannot identify whether the current branch is the last instruction with correct prediction or it is the second last branch while the last branch (at the same address) fails the prediction. 

That is, when the reported address is for a branch the valid combinations for decoder to determine last branch should be as follow:

-  correctly predicted: branch_count=1, branch_fmt=10
- mis-predicted: branch_count=0, branch_fmt=01

It is a pretty tricky corner case however.